JdeRobot/RoboticsAcademy #1: initial release of the repo

jmplaza, 2016-01-07:
I created the repo some time ago for Teaching Robotics, and added three days ago the right writting permissions and team on github. Now it is ready for action.
Now several components from JdeRobot must be moved to this repository: introrob, introrob_py, introrob_qt at least. FollowBall component too, though it is obsolete and needs update.

jmplaza, 2016-01-07:
It should the Cmake and CPack machinery, dependences, etc. in order to compile it when installed and the automatic generation of the corresponding debian package. The users will mainly install this through a debian package.

varhub, 2016-01-09:
I'm discourage it. CPack must not be done because it is planned for aptitude and "stable" and tested sources.
The only advantage to create a .deb is to embed cmake_uninstall into apt-get uninstall task so build directory can be safety removed.

But wait! 
Why an student *must* do **make && sudo make install** when just **make** and run from build directory is the faster way to go and is also compatible with IDEs?

Corollary: enforce a student to work with sudo may lead in broken SO and reinstall it from scratch is not a pleasant job.

closed 2016-02-03 by jmplaza