JdeRobot/RoboticsAcademy #32: replace introrob-qt with kobuki-viewer in follow_turtlebot exercise

jmplaza, 2016-02-23:
KobukiViewer is the standard tool in JdeRobot to teleoperate the Kobuki robot. Let us use it to move the turtlebot. This tool is included in the JdeRobot package, so no additional component is needed.
The text of the exercise description should be updated. In addition, the JavaScript version of KobukiViewer should be mentioned, it may also be used to teleoperate the turtlebot while the drone is pursuing it.

fqez, 2016-03-17:
KobukiViewer now includes a little label showing the current v and w sended to the robot in real time. It also has the laser widget fixed and shows the laser area changing in real time too. 
[Here's](http://jderobot.org/store/fperez/uploads/videos/absolute_interfaces.webm) a video showing how it looks like now.

almartinflorido, 2016-11-06:
This issue has been fixed in various commits:
https://github.com/JdeRobot/TeachingRobotics/issues/104
https://github.com/JdeRobot/TeachingRobotics/issues/99

closed 2016-11-06 by almartinflorido