JdeRobot/RoboticsAcademy #35: scenario for global navigation exercise

jmplaza, 2016-02-24:
The infrastructure needed is composed of a car model, a small city in Gazebo, and a png map of such city. 

The car model (taxi) will be included in JdeRobot repository, so it can be used in general. The Gazebo world file will include the town in 3D. The town map in png will be provided to the application software to use it for path planning.

reysam93, 2016-03-03:
The world and the map of the city have been added in the teleTaxi folder. Also the models, as I haven't made the pull request to JdeRobot yet.

fqez, 2016-03-03:
Hi, 
it's ok by now, but when the models are in the official repo could you please remove them from the exercise? This way we have all the models located in one directory (/usr/local/share/jderobot/gazebo/models) and here we have only the worlds that use them. Good job!!

reysam93, 2016-03-03:
No problem, I have already remove it from this repository and the are in the JdeRobot repository, with the other models of gazeboserver.

closed 2016-03-03 by reysam93