JdeRobot/RoboticsAcademy #39: Create new gazebo world with obstacles inside a circuit

eperdices, 2016-03-16:
Create a new gazebo world similar to "simpleCircuit.world" including obstacles (vehicles). This world will be used to test a VFF algorithm.

fqez, 2016-03-17:
I'm goint to add the simpleCircuitObstacles.world from the JdeRobot official repo. Could you tell me please in which example you want the world?
Thanks!

eperdices, 2016-03-28:
I am creating a new exercise called obstacle avoidance to implement a VFF algorithm, but I haven't uploaded it yet.

I have seen that you have already uploaded to JdeRobot a new world with obstacles, I will try to use that one and I will tell you if it works fine.

eperdices, 2016-03-28:
The new world works ok and has enough cars to test the algorithm, so I close this issue.
Thanks!

closed 2016-03-28 by eperdices