JdeRobot/RoboticsAcademy #40: Fake walls in simpleCircuit.world

eperdices, 2016-03-17:
As you can see on the picture there are two wall at both sides of F1 car, left wall is detected correctly with laser but right wall is "transparent" to laser measures. The same happens if I try to crash the car into the walls, left wall works well but the car is able to go through the right wall as if it didn't exist.

[image]

fqez, 2016-03-17:
Hi, yes the collision for that walls were not modeled. I didn't think it was necessary, but if you need I can add them to.

fqez, 2016-03-26:
Should be solved here: https://github.com/RoboticsURJC/JdeRobot/pull/395

eperdices, 2016-03-28:
Works perfect, thank you!

closed 2016-03-28 by eperdices