JdeRobot/RoboticsAcademy #41: [bug] follow_line worlds are broken

varhub, 2016-03-27:
# follow_line: worlds are broken

affected:
* circuit.world
* simpleCircuit.world

Gazebo hangs and requires force close. Console log:
```
Load: -1517115872 cam_f1_left
Load: -290466432 cam_f1_right
LaserDump Constructor
constructor motors
-----------------constructor Pose3D
GetName() Pose3D: f1
[Info] loaded Ice.Config file: /usr/local/share/jderobot/conf/f1Pose3D.cfg
Pose3D Endpoints default -h localhost -p 8998
--Ice.Config=f1Motors.cfg
[Info] loaded Ice.Config file: /usr/local/share/jderobot/conf/f1Motors.cfg
Motors Endpoints > default -h localhost -p 9999
[Wrn] [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
 laser: f1::laser
strs[0]: f1
[Info] loaded Ice.Config file: /usr/local/share/jderobot/conf/f1_laser.cfg
Laser Endpoints > default -h localhost -p 8996
[Msg] Waiting for model database update to complete...
```

Just **[Msg] Waiting for model database update to complete...** forever.

## PS: much better instructions to execute it are required. Even more important when it targets to novel students.

### Verbose info
Executing from gazebo directory:
```
lrwxrwxrwx 1 varribas varribas    19 Mar 27 15:09 cam_pioneer_left.cfg -> cam_sensor_left.cfg
lrwxrwxrwx 1 varribas varribas    20 Mar 27 15:09 cam_pioneer_right.cfg -> cam_sensor_right.cfg
-rw-rw-r-- 1 varribas varribas   225 Mar 26 14:51 cam_sensor_left.cfg
-rw-rw-r-- 1 varribas varribas   225 Mar 26 14:51 cam_sensor_right.cfg
-rw-rw-r-- 1 varribas varribas   406 Mar 26 14:51 circuit.world
-rw-rw-r-- 1 varribas varribas     0 Mar 26 14:51 default
-rw-rw-r-- 1 varribas varribas 67696 Mar 26 14:51 montmelo-pioneer.world
-rw-rw-r-- 1 varribas varribas    67 Mar 26 14:51 pioneer2dxEncoders.cfg
-rw-rw-r-- 1 varribas varribas    65 Mar 26 14:51 pioneer2dxMotors.cfg
-rw-rw-r-- 1 varribas varribas   234 Mar 26 14:51 pioneer2dx_pose3dencoders.cfg
-rw-rw-r-- 1 varribas varribas   412 Mar 26 14:51 simpleCircuit.world
```

fqez, 2016-03-27:
There are missing models in the .gazebo/models directory. Do you have the lastest f1 model installed? Works fine for me. It's a little heavy to load because the model is pretty big, but works fine for me. Make sure you have the the f1 model in the lastest version.

varhub, 2016-03-27:
Sorry bro,
You must known better than any other that missing models are recognized as:
```
[Wrn] [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
[Wrn] [gazebo.cc:186] Waited 1seconds for namespaces.
...
[Wrn] [gazebo.cc:186] Waited 1seconds for namespaces.
[Err] [gazebo.cc:191] Waited 11 seconds for namespaces. Giving up.
[Err] [Node.cc:98] No namespace found
...
[Err] [Node.cc:98] No namespace found
```

And eventually, after database timeout, hint if finally presented:
```
[Err] [ModelDatabase.cc:408] Unable to download model[model://pistaSimple]
[Err] [SystemPaths.cc:399] File or path does not exist[""]
Error [parser.cc:528] Unable to find uri[model://pistaSimple]
[Err] [ModelDatabase.cc:408] Unable to download model[model://f1]
[Err] [SystemPaths.cc:399] File or path does not exist[""]
Error [parser.cc:528] Unable to find uri[model://f1]

```

varhub, 2016-03-27:
JdeRobot version info:
```
(hJdeRobot5_up)varribas@Ubuntu14lts:~/curso_basico$ dpkg -l | grep jderobot
ii  jderobot                                  5.3.1-rc6                                           amd64        JdeRobot is a software development suite for robotics applications,
ii  jderobot-buildtools                       1.0.0                                               all          Developers toolkit for JdeRobot
ii  jderobot-deps-dev                         5.3.1                                               all          Metapackage that gathers all library headers required to compile JdeRobot framework.
ii  jderobot-teachingrobotics                 0.1.2                                               amd64        JdeRobot is a software development suite for robotics applications,
```

fqez, 2016-03-27:
Copy all the models from JdeRobot to .gazebo/models and try again.

fqez, 2016-03-28:
Also, the F1 model version that comes with the pacakage isn't the last version. So, the fixed model is in github and not in the rc6 package yet. That model was giving that problem because it was trying to load the hokuyo laser mesh from the turtlebot model and not from the f1 model. So if you download the F1 model from github and copy it to .gazebo/models it should work.

varhub, 2016-03-28:
Understood ^^

varhub, 2016-03-28:
Working except this:
```
[Err] [Material.cc:93] Unable to find texture[Color_006.jpg] in path[/home/varribas/curso_ws/.gazebo/models/pista/meshes]
```
Noticed that **lateral protections are red** and sometimes black-gray.
Is this the expected behavior?

fqez, 2016-03-28:
The black-gray parts are a lighting issue. In the 3D modeling software (sketchup/blender) it is all red, but the ambient factor of the texture is not good. I can solve it if it bothers you so much.

varhub, 2016-03-28:
Its not a problem except for color filter ;)
Let it there for make it a little harder.

closed 2016-04-22 by varhub