JdeRobot/RoboticsAcademy #59: new exercise: visual marker based 3D localization, host component

jmplaza, 2016-04-25:
The host component to this exercise should provide GUI to see each detected marker on images, and local relative 3D estimations, and absolute 3D estimations.

The solution has basically three steps:
1.- marker detection in 2D image (both Aruco or Apriltags)
2.- 3D pose estimation from each detected marker, using Perspective-n-Points in OpenCV.
3.- estimation fusion: several markers in the same image and several estimations along time (maybe Kalman filtering)

Alberto programmed a C++ solution for all this in [http://jderobot.org/Alopezceron-tfm]

trunc8, 2021-04-05:
@jmplaza is this exercise still in the pipeline? I have coded solutions for the three steps in individual assignments in the past. This sounds like an interesting problem. Is there any way to work on this as part of GSoC? (I have completed all 4 programming challenges)

jmplaza, 2021-04-05:
Hi @trunc8 , 

that exercise is in the pipeline, but not as part of GSoC-2021. For GSoC-2021 questions please use its own forum (https://forum.jderobot.org/c/english/gsoc-2021/14), and the candidate selection process we strictly follow (https://jderobot.github.io/activities/gsoc/2021#application-instructions-for-gsoc-2021), better than this repository. 

Right now useful contributions to RoboticsAcademy are, for instance, (a) betatesting all the exercises available in 2.3 release, (b) giving feedback through the RoboticsAcademy forum (https://forum.jderobot.org/c/english/roboticsacademy/12), (c) solving any open issue, etc.. But all of them are fully decoupled from GSoC-2021. Please don't mix topics and channels.

jmplaza, 2021-11-21:
superseded by #1297

closed 2021-11-21 by jmplaza