JdeRobot/RoboticsAcademy #67: the components should work when launched by their name, without python command

jmplaza, 2016-05-23:
from the command line. At the same time, related to #27, better names should be used instead of generic ones like 'main.py'. For instance follow-line.py, obstacle-avoidance.py....

eperdices, 2016-05-25:
Names and headers changed in these exercises:

latyrinth_escape
obstacle_avoidance 
rescue_people

I also had to change permissions to execute python code directly:
chmod a+x main.py

reysam93, 2016-06-03:
In global_navigation exercise the main.py is named globalNavigation.py and it has permissions to be executed directly.

almartinflorido, 2016-06-05:
Names and headers changed in these exercises:
color_filter
cat_py
follow_road
introrob_py
position_control
follow_turtlebot

Added README files to explain the basic configuration to launch the components.

chanfr, 2016-06-06:
Names and headers changed in these exercises:
follow_line
3d_reconstruction

closed 2016-06-06 by jmplaza