JdeRobot/RoboticsAcademy #101: Update introrob_py component to python 3 and qt5

almartinflorido, 2016-11-05:


jmplaza, 2016-11-05:
Maybe we can skip this issue, as introrob_py is no longer the central component for everything. It was, but now the contents of the Teaching Robotics are scattered along several academic components, mainly one per exercise. In fact, we could even delete from the repository. What do you think?

almartinflorido, 2016-11-05:
Ok, but then what does component we can use to teleoperate ardrone?.  We can use the other components in this repo but some of them are specify for a task.

In any case the changes has been made (minor changes). I'll commit the change if is necessary we can remove the component.

Ok?

jmplaza, 2016-11-05:
Perfect. 
To teleoperate the ardrone we can use uavviewer, the tool for drones in JdeRobot.

almartinflorido, 2016-11-05:
Done!

closed 2016-11-05 by almartinflorido