JdeRobot/RoboticsAcademy #110: JdeRobot logo in the GUI of all the academic applications

jmplaza, 2016-12-29:
Let's put the logo of our project (the four colored circles) in the GUI of all the academic applications. Let's start with: 
- [ ]  	3d_reconstruction
- [ ] 	bump_and_go 	
- [ ]	color_filter 	
- [ ]	drone_cat_mouse 	
- [ ]	follow_line 	
- [ ]	follow_road 	
- [ ]	follow_turtlebot 	
- [ ]	global_navigation 	
- [ ]	labyrinth_escape 	
- [ ]	obstacle_avoidance 	
- [ ]	position_control 	
- [ ]	rescue_people 	
- [ ]	visual_lander

aitormf, 2017-01-02:
first approach in follow_line, creating a widget:

[image]

jmplaza, 2017-01-02:
Great Aitor

closed 2018-04-13 by aitormf