JdeRobot/RoboticsAcademy #188: prepare a solution for Bump&Go exercise using VisualStates

jmplaza, 2018-03-22:


ShivamSinghania, 2018-03-25:
Does this include both bump_and_go_py and bump_and_go ? or just one of them ? Can I work on this or is this for you ?

jmplaza, 2018-03-26:
Hi @ShivamSinghania, before working on this issue, I suggest you better use and solve some JdeRobot-Academy exercises. That would give you better understanding of this teaching framework. Maybe the [color filter exercise](https://github.com/JdeRobot/Academy/tree/master/src/color_filter) and the [drone cat-mouse exercise](https://github.com/JdeRobot/Academy/tree/master/src/drone_cat_mouse) are good points to start.

In addition, running the [tutorials about VisualStates](https://jderobot.org/Tutorials#VisualStates_tool) will give you more context. 

Beta testing all of them and refining them are also good contributions.

pushkalkatara, 2018-04-07:
Can I work on this issue? @jmplaza . I would create 3 states Go, GoBack and Rotate as in this [video](http://jderobot.org/store/s.rey/uploads/videos/bumpAndGo.ogv) and an XML to load the automata with three states with their state and transition code and aux code as in obstacleAvoidance example.

jmplaza, 2018-04-08:
Sure @pushkalkatara. You have some experience with VisualStates. Again I suggest you better use and solve some JdeRobot-Academy exercises before. That would give you better understanding of this teaching framework. Maybe the color filter exercise and the drone cat-mouse exercise are good points to start.

This exercise is somehow different to the others, as it does not require any host academic node where the student has to put her code. This bump&go exercise has to be solved directly using the [VisualStates tool ](https://github.com/JdeRobot/VisualStates) to generate the node with the robot intelligence. Not the VisualState in this [local directory](https://github.com/JdeRobot/Academy/tree/master/src/bump_and_go_py/visualStates_py), which will be deleted soon.

pushkalkatara, 2018-04-11:
@jmplaza I developed the xml for bump and go behavior in VisualStates with the help of Kobuki obstacleAvoidance example using Ice interface.
[image]
The output of the same in Gazebo is - 
[image]
I think fine tuning the auxiliary code is required according to the BumpAndGo behavior. 
 Currently in Go state,
```
myMotors.sendV(0.4)
myMotors.sendW(0)
```
GoBackState
```
myMotors.sendV(-0.4)
myMotors.sendW(0)
```
Rotate state
```
myMotors.sendV(0)
myMotors.sendW(0.1)
```
and the conditions are same as the obstacleAvoidance tutorial.
Please help me with a case scenario or online tutorial according to which the auxiliary code should be changed. In this [video](http://jderobot.org/store/s.rey/uploads/videos/bumpAndGo.ogv) there is a variable distance which is being checked for less than zero, according to it the back and rotate states are being activated. Further, the angle, destiny and the speed is being manipulated through checking the laser data. Is this manipulation correct?

jmplaza, 2018-04-13:
Hi @pushkalkatara, @okanasik,

you better test the ROS interface release of the exercise, it is quite similar. The video you mention is quite old. You can check the most recent tutorial on VisualStates [here](https://jderobot.org/Tutorials#TurtleBot_Example_with_ROS)

Maybe you can solve this exercise with three states instead of two:
(1) State MoveForward, having an obstacle in front closer than 1m can be the transition to jump to the next state.
(2) State GetBack, staying on this state for 2 seconds is a good way to do it (this will let you play with Temporal transitions)
(3) State Rotate, instead of using a Temporal Transition again you can get the current robot orientation when entering in this state, then pick a random angle ([-PI,+PI]) and add them. That sum sets the end-of-rotation orientation. On each iteration of the automaton on this state the current orientation should be checked: in case of no reaching the end-of-rotation orientation the robot should keep rotating; in case of that orientation reached the automaton should change to "MoveForward" state again. For programming this you will have to use functions and variables at VisualStates. They both form the auxiliary code.

You can check the [README file](https://github.com/JdeRobot/Academy/tree/master/src/bump_and_go_py) corresponding of this exercise for further directions.

Cheers,

pushkalkatara, 2018-04-15:
@jmplaza @okanasik Please review the BumpAndGo behavior. I made the necessary changes. Now the exercise is through ROS interface. 
1. MoveForward has laser threshold of 0.5
2. GoBack with temporal transition with wait for 2 seconds.
3. Rotate with conditional transition with condition if random_angle < current_yaw. I calculated yaw  using the orientation from /odom, converted tbhe orientation to euler form using tf library.  
Here's the link to the ROS workspace - 
[VisualStatesWS.zip](https://github.com/JdeRobot/Academy/files/1911694/VisualStatesWS.zip)

okanasik, 2018-04-15:
Dear @pushkalkatara , I checked your project. It achieves what is expected. Just a small change, you may change the name of 2 second temporal transition to GetBack2Sec instead of GetBack3sec :) Also, maybe it could be better if you can name the transitions according to the their functions. For example, RotateRandom transition could be named as Rotated.

Best,

pushkalkatara, 2018-04-15:
Thanks for the review @okanasik. I made the necessary changes. Should I make a pull request as an example in JdeRobot directory?

pushkalkatara, 2018-04-16:
Please close this issue as it is referenced to https://github.com/JdeRobot/VisualStates/issues/14

closed 2020-06-08 by pariaspe