JdeRobot/RoboticsAcademy #261: Error while installing electron

preeti98, 2018-10-07:
I am new to this community and have successfully installed JdeRobot in a virtual-box as I am using Ubuntu 18.04 currently. However I am getting the following error while installing electron under 3D-reconstruction exercise:
```
preeti@preeti-VirtualBox:~/Academy/exercises/3d_reconstruction/3DVizWeb$ npm install electron --save-dev --save-exact

> electron@1.8.8 postinstall /home/preeti/Academy/exercises/3d_reconstruction/3DVizWeb/node_modules/electron
> node install.js

sh: 1: node: not found
3DViz@0.1.0 /home/preeti/Academy/exercises/3d_reconstruction/3DVizWeb
`-- jquery@3.3.1  extraneous

npm WARN 3DViz@0.1.0 No description
npm WARN 3DViz@0.1.0 No repository field.
npm WARN 3DViz@0.1.0 No license field.
npm ERR! Linux 4.15.0-36-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "electron" "--save-dev" "--save-exact"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! electron@1.8.8 postinstall: `node install.js`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the electron@1.8.8 postinstall script 'node install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/preeti/Academy/exercises/3d_reconstruction/3DVizWeb/npm-debug.log
```

vinay0410, 2018-10-08:
It's probably because the node command wasn't found on shell.
Could you try running the following commands in your virtual box Ubuntu.

```
echo "alias node='nodejs'" >> ~/.bashrc
source ~/.bashrc
```

See if that works for you.

preeti98, 2018-10-08:
I tried the above set  of commands and then I used the following inside /Academy/exercises/3d_reconstruction/3DVizWeb folder
```
npm install electron --save-dev --save-exact

```
But I  still get the same error message  as above.

vinay0410, 2018-10-08:
Your issue is because the command called node isn't found.
So, try running ```sudo ln -s /usr/bin/nodejs /usr/bin/node```.
Now, node will be found.
if **not** then only run ```sudo apt install nodejs-legacy```.

Once this is resolved, you might face an error regarding downloading electron, if that occurs update npm by running, ```sudo npm install -g npm```.

preeti98, 2018-10-11:
Ok, thank you! I will look into the exercises you mentioned and will look into 3D Reconstruction later.

jmplaza, 2020-05-03:
3D reconstruction exercise has been refactored and no longer depends on 3DVizWeb, now it shows the 3D point cloud inside the PyQt window.

closed 2020-05-03 by jmplaza