JdeRobot/RoboticsAcademy #279: Couldn't find an AF_INET address for []

Aman-py, 2019-03-16:
when i am trying to running:
python ./follow_line.py follow_line_conf.yml
nothing is showing just appears another error

Couldn't find an AF_INET address for [] 

in another shell where this command is running:
roslaunch /opt/jderobot/share/jderobot/launch/f1.launch

and getting screen like this:

[image]

pankhurivanjani, 2019-03-17:
Hello!
For AF_INET, this error is regarding ROS-IP.
Check into this link
https://answers.ros.org/question/163556/how-to-solve-couldnt-find-an-af_inet-address-for-problem/

I would also recommend checking kobuki drivers, if they are installed or not. Locate the kobuki-simple-ros.launch file.

Aman-py, 2019-03-17:
@pankhurivanjani  I have tried solution that is provided by this link problem is regarding to AF_INET error for local machine and link provides solution for other hosts machine.
i have installed all kobuki drivers they are installed problem is to solve is how to provide AF_INET address for local machine we have to set it using export command and i want to know that

pankhurivanjani, 2019-03-18:
Hi @Aman-py 
Okay, So the issue is regarding ROS-IP on the local machine.
I looked into the error more precisely.
Follow this link, it might help you.
https://youtu.be/UsuQwUkzNII

jmplaza, 2021-03-29:
This is no longer required as all the dependencies of the exercises are pre-installed in the RADI since RoboticsAcademy 2.3, and they work fine, including the FollowLine exercise.

closed 2021-03-29 by jmplaza