JdeRobot/RoboticsAcademy #286: Model world for exercises not loading in Gazebo

pankhurivanjani, 2019-03-17:
I am trying to execute exercises, but after launching the 'launch' file, Gazebo environment is opening but the world model isn't coming into view.
I checked the world file arguments inside the .launch file and entered the world file along with the launch file manually, still getting the same result.
[image]

fqez, 2019-03-17:
Hi @pankhurivanjani 
did you source the gazebo setup?

```
$ source /opt/jderobot/share/jderobot/gazebo/gazebo-setup.sh
$ source /opt/jderobot/share/jderobot/gazebo/gazebo-assets-setup.sh
```

if so, try to activate the verbose flag in the launch file (change it from false to true) and post here the output of the roslaunch command

Regards!

pankhurivanjani, 2019-03-18:
Hello @fqez 
I had not setup gazebo-assets-setup.sh.
Now, output of my roslaunch command is as follows:
... logging to /home/pankhuri/.ros/log/35242e18-494a-11e9-896b-1c3947c53712/roslaunch-pankhuri-LENOVO-IDEAPAD-500-15ISK-12695.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pankhuri-LENOVO-IDEAPAD-500-15ISK:46021/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [12714]
process[gazebo_gui-2]: started with pid [12719]
Gazebo multi-robot simulator, version 7.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1552892173.436870885]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1552892173.437704710]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.22.32.34
[ INFO] [1552892173.473376887]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1552892173.474439951]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.22.32.34
[Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
[Err] [ModelDatabase.cc:414] Unable to download model[model://pistaSimple]
[Err] [SystemPaths.cc:429] File or path does not exist[""]
Error [parser.cc:689] Unable to find uri[model://pistaSimple]
[Err] [Material.cc:100] Unable to find texture[generic_main.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[glass.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[re0_wheel.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[re0_main.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[tyre.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[tread_slick.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[light_glow.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[ INFO] [1552892323.351735251]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1552892323.362877204]: PlanarMovePlugin missing <robotNamespace>, defaults to ""
[ INFO] [1552892323.362933704]: PlanarMovePlugin missing <enableYAxis>, defaults to "1"
[ INFO] [1552892323.363846861]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materia[ INFO] [1552892323.448337542, 0.026000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
ls/textures/generic_main.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/generic_main.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/glass.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/glass.png]
[ INFO] [1552892323.468552400, 0.040000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_wheel.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_wheel.png]
[... 29 lines cut ...]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/light_glow.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/light_glow.png]
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.

************************************************************************************************
Regarding textures in /opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/
they are indeed missing:

root@pankhuri-LENOVO-IDEAPAD-500-15ISK:/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault# ls
Car.lwo  Car.mtl  Car.obj  materials  Wheel.lwo  Wheel.mtl  Wheel.obj

but they are here,
root@pankhuri-LENOVO-IDEAPAD-500-15ISK:/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/materials/textures# ls
generic_main_nm.png       int_info_pod.png    tread_slick.png
[... cut ...]
glass.png                 light_glow.png      tyre.png
int_digi_placeholder.png  re0_main.png        tyre_specocc.png
int_digi_tach.png         re0_wheel.png
int_indicator.png         tread_slick_nm.png

Changing the path is not an issue.
In f1.launch file, world was 'fi-simple-circuit.world', but in world files in my system, I found f1_1_simpleciruit.world, So I replaced it.
********************************************************
Should I begin by changing paths in file or is there some other alternative?

kurshakuz, 2019-03-19:
Hello everyone, I am having the same issue. I have both sourced gazebo-setup.sh and gazebo-assets-setup.sh but my world is not loading. I tried to open internal gazebo7 worlds and it worked fine. It is not ROS supported exercise so I cannot see the launch files as suggested above. Any ideas on what should help?

[image]

fqez, 2019-03-19:
About @pankhurivanjani problem, I think your issue is that you are launching the exercise from /opt/jderobot/share/jderobot/launch/f1.launch, but you should launch it from /opt/jderobot/share/jderobot/**gazebo**/launch/f1_1_simplecircuit.launch
The first launch directory is meant to launch drivers not related with gazebo but ROS. Try this way and let me know if you succeed.

About @kurshakuz , after sourcing gazebo setups, try going to /opt/jderobot/share/jderobot/**gazebo**/launch and execute
```
roslaunch car_1_junction.launch
```

Sorry guys about the documentation of this being out of date, I will try to update and improve launch recipes asap.

kurshakuz, 2019-03-19:
Thank you very much, @fqez I am finally able to launch it! As I have find out, the only exercises that are  not ROS integrated right now are only the _car_junction_ and _car_autoparking_, but following your instructions I now see that these exercises are currently being moved to ROS (I see that these exercises already have necessary rostopics and rosservices). I wanted to know what is the current stage of development, and is there something right now I can help you or anyone working on it? 
I have played around with _car_junction_ world a little bit with ROS messages and if needed can start helping with development of its exercise.

pankhurivanjani, 2019-03-20:
Thank you @fqez ,after launching it from /**gazebo**/launch I was able to get the car, but the whole world is still not there and it has the same issues as I quoted previously.
Here is my gazebo environment:
[image]

My errors are : 
[image]

Looking into errors:
1. ERROR-1
 [Err] [ModelDatabase.cc:414] Unable to download model[model://pistaSimple]
[Err] [SystemPaths.cc:429] File or path does not exist[""]

After checking into opt/jderobot/share/jderobot/gazebo/models there is file named pista_simple, but no file named pistaSimple

2. ERROR-2

[Err] [Material.cc:100] Unable to find texture[generic_main.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[glass.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[re0_wheel.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[re0_main.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[tyre.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[tread_slick.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]
[Err] [Material.cc:100] Unable to find texture[light_glow.png] in path[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/]

After checking the files,

[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault/materials/textures]

3. [Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/generic_main.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/generic_main.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/glass.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/glass.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_wheel.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_wheel.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_main.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/re0_main.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/tyre.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/tyre.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/tread_slick.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/tread_slick.png]
[Err] [SystemPaths.cc:429] File or path does not exist["/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/light_glow.png"]
[Err] [RenderEngine.cc:484] URI doesn't exist[/opt/jderobot/share/jderobot/gazebo/models/f1_renault/Renault//../materials/textures/light_glow.png]

What I understood  is that I need to change the path in material.cc and change nameof pistasimple.

abhiMishra98, 2019-03-22:
I am facing the same problem. The model world, after launching the launch file is not getting loaded. Please help. i have tried sourcing gazebo files and the gazebo assets but none helped me.

pankhurivanjani, 2019-03-23:
Hi,  @fqez  I copied the textures in f1_renault/Renault and was able to get rid of few errors.
But one error is still there,
[Err] [ModelDatabase.cc:414] Unable to download model[model://pistaSimple]
[Err] [SystemPaths.cc:429] File or path does not exist[""]
Error [parser.cc:689] Unable to find uri[model://pistaSimple

I have figured out by checking files that we need to change pistaSimple to pista_simple in ModelDatabase.cc , But I am not able to find this file.

[image]

fqez, 2019-03-23:
Hi, nicely done! the pista_simple model is kinda new so I doubt it was on the assets package. Try upgrading the jderobot-gazebo-assets debian package because last Thursday I released a new version with all that new models included.
If you don't want to do so, you can clone the assets repository (https://github.com/jderobot/assets) and get the model from there. It should be installed in /opt/jderobot/share/jderobot/gazebo/models

Hope it helps.

pankhurivanjani, 2019-03-24:
It is working fine now!!
Thanks a lot @fqez 
Also, I have opened a new pull request in [#09]https://github.com/JdeRobot/RoboticsAcademy/pull/289l) where I have update ReadMe for launching this exercise.
I hope this will help other members too! :)

closed 2019-03-24 by pankhurivanjani

PaulNicolasHunter, 2019-04-05:
@fqez  even though I'm not getting any errors, my camera output is not showing anything, it's black.

[image]

and what's up with the docker exec? when I do `docker run -tiP --rm -p 7681:7681 jderobot/jderobot world flyingkinect_1_simple.world` it gives me 

`/jderobot_entrypoint.sh: line 16: exec: world: not found
`
any help please?