JdeRobot/RoboticsAcademy #299: Unable to execute Docker image.

PaulNicolasHunter, 2019-04-04:
I'm trying to follow the exercises in the **RobotocsAcademy** (follow_road to be specific).
I'm getting problem running the **world** exec as stated in the Readme.

here is the screen shot of it.

[image]

I also looked at the JdeRobot's docker directory, couldn't find anything with **world**, 
what I'm missing here?

Pravin854, 2019-04-05:
Did you solve it ? I'm also getting same error for `sudo docker run -tiP --rm -p 7681:7681 jderobot/jderobot world  quadrotor_1_roadtextures.world`

PaulNicolasHunter, 2019-04-06:
@Pravin854 nope, but I tried to follow their instructions in [their docker profile](https://hub.docker.com/r/jderobot/jderobot/), but still no luck, 

[image]

jmplaza, 2021-03-29:
Current RADI (docker image) runs fine with all the dependencies of FollowLine exercise and the Gazebo world is launched without problems.

closed 2021-03-29 by jmplaza