JdeRobot/RoboticsAcademy #300: Update Readme for Academy

avijitashe, 2019-04-05:
I'd faced issues with installing Jde Robot Academy while completing GSoC exercises. I tried directly by following the Readme here.. The jderobot_gazebo_assets are not found by apt-cache search, for example. It also gives failed dependency issues. Are these some common issues or this installation instructions are not complete?

I'm using a fresh Ubuntu 16 and ROS Kinetic machine. The gazebo is 14.X. All these are inside an Oracle VM. Thank you!

pankhurivanjani, 2019-04-06:
Have you tried this?
sudo apt install jderobot-gazebo-assets

Also for dependencies have you done this?
sudo apt install jderobot-deps-dev

Pravin854, 2019-04-06:
Hey @avijitashe, if you still get the error then follow the steps mentioned in https://jderobot.org/Installation

SakshayMahna, 2019-10-03:
The webpage [https://jderobot.org/Installation](https://jderobot.org/Installation) does not exist.

Some additional instructions need to be performed before successfully installing the gazebo dependency. I can resolve the issue by updating the README.md

jmplaza, 2019-10-05:
Yes, we are in the process of updating documentation. @SakshayMahna has already updated the REAME.md properly. Now it has a working installation recipe. Check also the new [RoboticsAcademy webpage](https://jderobot.github.io/RoboticsAcademy/) based on GitHub Pages.

closed 2019-10-05 by jmplaza