JdeRobot/RoboticsAcademy #301: roslaunch f1_1_simplecircuit.launch is empty

avijitashe, 2019-04-05:
When trying running the exercises for GSoC 19, for the first time, I get two errors often

1. An empty Gazebo world. There are no warnings or msgs. Upon looking at echo $GAZEBO_RESOURCE_PATH I get this /usr/share/gazebo-7:::

2. [Err] [Server.cc:383] Could not open file[amazonrobot_1_warehouse.world]
[Wrn] [Server.cc:342] Falling back on worlds/empty.world
This is with another launch file, not f1_circuit

I guess I'm missing some instructions for the FIRST run of the package. Kindly guide me. Thanks!

pankhurivanjani, 2019-04-06:
Hi @avijitashe 
Have you sourced the gazebo environment?
Also, I would recommend you to go through the discussion on this issue [https://github.com/JdeRobot/RoboticsAcademy/issues/286](#Issue286) 

This might solve your issue.

igarag, 2020-01-19:
Hi! Thanks for the answer Pankhuri.

Was this incident resolved @avijitashe? Can I close it?

avijitashe, 2020-01-20:
The answer was not helpful in my specific scenario, but I solved it some
other way back then. You may CLOSE it.

jmplaza, 2020-01-20:
Glad to hear that @avijitashe. We have just refactored the Academy web page (the installation instructions require a little update though). Please feel free to play with other ready-to-use exercises, they are available [here](https://jderobot.github.io/RoboticsAcademy/exercises/)

closed 2020-01-20 by jmplaza