JdeRobot/RoboticsAcademy #343: [GitHub Pages] - News section

igarag, 2019-09-02:
The `navbar` can be a good place to put it.

igarag, 2020-02-03:
Maybe in left nav bar?

trunc8, 2021-04-02:
Hi @igarag, is this issue still open? If yes, please could you provide a description of what you have in mind?

igarag, 2021-04-05:
Hello @trunc8.

The idea with this issue is to create a news section decoupled from the main page. A simple idea might be to include a new button at the top, called "news", that links to the content that is currently on the landing page.

If @jmplaza agrees, we can implement it, if you both agree.

Regards.

jmplaza, 2021-04-05:
I think we have other more important issues. Now we have a [RoboticsAcademy forum](https://forum.jderobot.org/c/english/roboticsacademy/12) and a [roadmap page](https://jderobot.github.io/RoboticsAcademy/roadmap/), more than enough for users and developers to keep updated about this project... I would close this issue.

closed 2021-06-23 by jmplaza