JdeRobot/RoboticsAcademy #346: Installing requirement for Follow Road Exercise

luiscaiza, 2019-09-19:
Hello everybody,

I am trying to run Follow Road, the recepi tells to  install the jderobot-assets, dronewrapper and rqt_drone_teleop packages, but I can't and the terminal shows:
luis@lc-desktop:~$ sudo apt-get install ros-kinetic-drone-wrapper ros-kinetic-rqt-drone-teleop ros-kinetic-jderobot-assets
[sudo] password for luis: 
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
E: No se ha podido localizar el paquete ros-kinetic-drone-wrapper
E: No se ha podido localizar el paquete ros-kinetic-rqt-drone-teleop
E: No se ha podido localizar el paquete ros-kinetic-jderobot-assets

someone can help me? what is wrong?

nkhedekar, 2019-09-20:
Hi @luiscaiza, do you have ROS installed on your system? If not, you can install it by following the instructions here: http://wiki.ros.org/kinetic/Installation

diegomrt, 2019-09-24:
Hi @luiscaiza. If you installed ROS a few months ago and now you are having troubles with ``sudo apt update`` our ``sudo apt install``, they could be related to the change in the ROS GPG keys made to deal with [this security issue ](https://discourse.ros.org/t/security-issue-on-ros-build-farm/9342) happened on the ROS build farm last May. 

Please update your old key following the instructions of the link below. Once updated, run ```sudo apt update``` and after that ``sudo apt-get install ros-kinetic-nameofthepackage`` should work again.

[https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454](https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454)

jmplaza, 2021-03-29:
FollowLine exercise runs fine in the 2.3 release of RoboticsAcademy. All the required dependencies are already pre-installed in the RADI.

closed 2021-03-29 by jmplaza