JdeRobot/RoboticsAcademy #347: Trouble in installation

tanmayJ527, 2019-10-04:
I'm following the installation steps given in the README. 
In the final step
`sudo apt install jderobot-gazebo-assets`
I get the following error
`E: Unable to locate package jderobot-gazebo-assets`

Using Ubuntu 16.04

SakshayMahna, 2019-10-04:
Add the JDe Robot sources before using the install command. Maybe, due to a change in website, we have to set a new source.
```bash
sudo sh -c 'cat<<EOF>/etc/apt/sources.list.d/jderobot.list
# for ubuntu 16.04 LTS (64 bit)
deb [arch=amd64] http://wiki.jderobot.org/apt bionic main
EOF
```
```bash
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 24E521A4
```

Now, update the repositories
```bash
sudo apt update
```

Now, run
```bash
sudo apt install jderobot-gazebo-assets
```

jmplaza, 2019-10-05:
Solved in #348 Thanks @SakshayMahna !!

closed 2019-10-05 by jmplaza