JdeRobot/RoboticsAcademy #349: Import Error: No module named jderobot

tanmayJ527, 2019-10-08:
Running `python2 ./follow_line.py follow_line_conf.yml` gives 

`Traceback (most recent call last):`
`  File "./follow_line.py", line 31, in <module>`
`    from MyAlgorithm import MyAlgorithm`
`  File "/home/tanmay/Academy/exercises/follow_line/MyAlgorithm.py", line 8, in <module>`
`    import jderobot`
`ImportError: No module named jderobot
`.

tanmayJ527, 2019-10-09:
The `/opt/jderobot` directory does not have the file `setup.bash`. All it contains is a directory called `share`. Does this have something to do with the import error? 

`/opt/jderobot/share/jderobot` contains the directory `gazebo`.

SakshayMahna, 2019-11-28:
There is a little hack you can try!
For the follow_line exercise, the `MyAlgorithm.py` contains a line `import jderobot`. The imported library in this case serves no purpose, and you can remove it to run this exercise.

While, for some other exercises `jderobot` library is a requirement. Try following the instructions given on the [base](https://github.com/JdeRobot/base) repository. The line,

`sudo apt install jderobot`

may do the trick.  Also, you need to have **ROS Melodic(Ubuntu 18.04)** to fully install the framework. The above command may give dependency issues when tried to install on any other ROS.

jmplaza, 2021-03-01:
Jderobot-base or jderobot-assets are no longer a requirement for exercises in RoboticsAcademy 2.1 and 2.3, including drone related exercises

closed 2021-03-01 by jmplaza