JdeRobot/RoboticsAcademy #376: [Vacuum Cleaner] exercise vacuum_cleaner.launch giving error

SakshayMahna, 2019-12-31:
On running the command `roslaunch vacuum_cleaner.launch`, we receive the following error.
```bash
[Err] [Server.cc:379] Could not open file[Vacuum.world]
[Wrn] [Server.cc:338] Falling back on worlds/empty.world
[ INFO] [1577769279.565093917, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1577769279.645832462, 0.100000000]: Physics dynamic reconfigure ready.
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/opt/ros/melodic/share/turtlebot3_gazebo/models/turtlebot3_autorace"
[Err] [REST.cc:205] Error in REST request

```

SakshayMahna, 2019-12-31:
There are also some minor improvements required:

- Some spelling corrections
- Properly terminating `vacuumCleaner.py`

jmplaza, 2020-01-13:
Fixed in #377

closed 2020-01-13 by jmplaza