JdeRobot/RoboticsAcademy #381: [documentation]- change kinetic to melodic

mohamedsayed18, 2020-01-05:
this line 
sudo apt-get install ros-kinetic-desktop-full
need to be changed to this 
sudo apt-get install ros-melodic-desktop-full
in this link https://jderobot.github.io/RoboticsAcademy/installation/
I don't know if my query is valid to be posted here or should be discussed first in the mailing list

jmplaza, 2020-01-05:
Hi @mohamedsayed18, 

this is the right place to discuss that. Yes that should be changed, as ROS Melodic is the current underlying infrastructure for RoboticsAcademy. Feel free to submit a pull-request for this issue, but only after you have successfully tested ROS Melodic with one of the current operating exercises: [https://jderobot.github.io/RoboticsAcademy/portfolio/follow_line](https://jderobot.github.io/RoboticsAcademy/portfolio/follow_line) or [https://jderobot.github.io/RoboticsAcademy/portfolio/obstacle_avoidance/](https://jderobot.github.io/RoboticsAcademy/portfolio/obstacle_avoidance/)

You don't need to solve them, just to check that they are properly launched using ROS Melodic.

Cheers!

mohamedsayed18, 2020-01-06:
I tried the follow_line example and it worked fine.
I forked the jderobot.github.io but I don't know much about web technologies so I don't know which file is related to this link https://jderobot.github.io/RoboticsAcademy/installation/

SakshayMahna, 2020-01-06:
Jekyll framework along with Minimal Mistakes theme is the web technology used. You may find the tutorial online. All the files related to the webpage are present in the `docs` folder of the repository.

mohamedsayed18, 2020-01-07:
thanks @SakshayMahna I will check jekyll edit the file and send a pull request

jmplaza, 2020-01-07:
Hi @mohamedsayed18,

you have a pretty straightforward recipe here [https://github.com/JdeRobot/jderobot.github.io](https://github.com/JdeRobot/jderobot.github.io) in the README.md file about how to install Jekyll locally and then test your changes on the webpage before submitting your pull-request

In Academy we use the same Jekyll framework and Minimal Mistakes theme that we use for the general JdeRobot organization web page, which is stored in https://github.com/JdeRobot/jderobot.github.io

Cheers

jmplaza, 2020-01-07:
Solved in #384 Thanks @mohamedsayed18 !

closed 2020-01-07 by jmplaza