JdeRobot/RoboticsAcademy #389: Update of Amazon warehouse exercise from ROS-KInetic to ROS-Melodic

jmplaza, 2020-01-12:


jiban-jyoti, 2020-01-21:
Hello @jmplaza. Sir, I am facing some errors while running bump and go exercise. I had opened an issue regarding that but parallely can I work on Amazon warehouse exercise.

jmplaza, 2020-01-21:
Hi @jiban-jyoti , @kurshakuz is working on that exercise...

kurshakuz, 2020-01-24:
Dear @jmplaza, I have been working on the issue and found that when I installed jderobot package from scratch, updates that I've made [1] to assets repository haven't appeared in /opt/jderobot/share/jderobot/gazebo/models. Can you please clarify how should I add these updates for release?

[1] https://github.com/JdeRobot/assets/tree/master/gazebo/models/amazon_warehouse_robot

kurshakuz, 2020-01-24:
Updates that have to be made to the exercise:
1. cmd_vel_mux package is not maintained and therefore there is no release for Melodic. I'll write additional instructions on how to change between manual (teleoperation) and autonomous (move_base) movement by hand.
2. New Melodic source files accept frames in particular reference format in yaml config files for move_base for map and hokyo sensor frames. I'll made slight updates
3. Final configured .sdf model doesn't appear on release package, updates required

Apart from the stated above, everything works fine with reference solution

kurshakuz, 2020-02-01:
solved in https://github.com/JdeRobot/RoboticsAcademy/pull/421

closed 2020-02-01 by kurshakuz