JdeRobot/RoboticsAcademy #470: Installation Issue

Naman-ntc, 2020-03-09:
Hi, 
I have been trying to setup RoboticsAcademy on my system and am facing issues. I followed instructions on [this webpage](https://jderobot.github.io/RoboticsAcademy/installation/). 
But when I execute `roslaunch` or `colorTuner` I get `/bin/bash: command not found`. I have also tried sourcing my bashrc. 

To show the output of my system, I built a Colab file that mimics output at my system. Tou can find the colab [here](https://colab.research.google.com/drive/1doqk3GiX7eEqKH9hx1Q4KozyAfA7Lzfk). 
* In cell **5** output of `tail ~/.bashrc` contains `source /opt/ros/kinetic/setup.sh`. There is no such folder as `/opt/ros/kinetic` (`ls /opt/ros` yeilds on `melodic` folder in cell **8**). 
* In the next 2 cells (cell **6** and **7**), both `roslaunch` and `colorTuner` are found by the command line.

Can you please let me know how can I install it correctly and what mistake I am doing?

jmplaza, 2020-03-09:
Hi @Naman-ntc,

which exercise are you trying to run? Please try to work only with up-and-running exercises, which are listed on [this webpage](http://jderobot.github.io/RoboticsAcademy/exercises/). There are more exercises in the JdeRobot Academy source repository, but those not listed there are not currently ready to use.

At that webpage there is no reference to kinetic at all... Currently reference infrastructure for JdeRobot Academy is Ubuntu 18.04, ROS Melodic (not Kinetic) and Gazebo9

Naman-ntc, 2020-03-09:
Thanks for the quick response. I tried to run the follow_line experiment and color_filter experiment (latter is probably still under development?)

In followline exercise 
* it uses `roslaunch` which I am not able to execute on my setup with ubuntu 18.04 (command not found!)
* there is instruction to `source /opt/jderobot/share/jderobot/gazebo/gazebo-assets-setup.sh`. 
the output of `ls /opt/jderobot/share/jderobot/gazebo` **does not contain** `gazebo-assests-setup.sh`. I believe that script gets downloaded using `sudo apt installjderobot-gazebo-assets`.

SakshayMahna, 2020-03-09:
For the `roslaunch` to work correctly, try running the command `source /opt/ros/melodic/setup.bash`

Also, try the command you suggested, `sudo apt install jderobot-gazebo-assets`. Do let us know if this works!

Naman-ntc, 2020-03-10:
Hi, 
Thanks, yes it is working now.
Shouldn't the `jderobot-gazebo-assets` instruction be added on the installation webpage (in additional resources I presume)? 

Also should [post installation script](https://github.com/JdeRobot/base/blob/bef159bb6d9ab953815b89df775fa18dd63e1409/scripts/cmake/postinst) be updated to append `melodic/setup.bash` instead of `kinetic/setup.bash`?

jmplaza, 2020-03-10:
Hi @Naman-ntc , 

AFAIK the jderobot-gazebo-assets instruction is already included in the [installation webpage](https://jderobot.github.io/RoboticsAcademy/installation/#jderobot-assets). Isn't it?

[Color-filter](https://github.com/JdeRobot/RoboticsAcademy/tree/master/exercises/color_filter) exercise is not fully supported yet, and so the README is outdated. We are working on it.

Yes, updating the post installation script it would be useful. I have just creating the [corresponding issue](https://github.com/JdeRobot/base/issues/1393) in the JdeRobot/base repository. 

Thanks for your feedback!

Naman-ntc, 2020-03-10:
The installation instructions only mentions the following: 
`sudo apt install jderobot-assets` and `sudo apt install ros-melodic-jderobot-assets`.

From my understanding, `jderobot-gazebo-assets` has to be installed separately.

jmplaza, 2020-03-10:
"jderobot-assets" is the new package name for the old "jderobot-gazebo-assets" debian package. 

Could you please verify that a clean install using only 
`sudo apt install jderobot-assets and sudo apt install ros-melodic-jderobot-assets`
and 
`source /opt/ros/melodic/setup.bash`
works for you?

@fqez @pariaspe

Naman-ntc, 2020-03-10:
I can verify that on running 
`sudo apt install jderobot-assets` and `sudo apt install ros-melodic-jderobot-assets` 
along with
`source /opt/ros/melodic/setup.bash` and `source /opt/jderobot/setup.bash` 
does not install `gazebo-assets-setup.sh`.

You can check [the colab](https://colab.research.google.com/drive/1doqk3GiX7eEqKH9hx1Q4KozyAfA7Lzfk) to confirm too.

vikaskumareinstein, 2020-03-13:
I completely agree with @Naman-ntc .After performing `sudo apt install jderobot-assets` and `sudo apt install ros-melodic-jderobot-assets` alongwith the sourcing ,I am unable to find the `gazebo-assests-setup.sh` script.On that place I simply sourced the `assets-setup.sh` which doesnot work at all.

pariaspe, 2020-03-13:
> I completely agree with @Naman-ntc .After performing `sudo apt install jderobot-assets` and `sudo apt install ros-melodic-jderobot-assets` alongwith the sourcing ,I am unable to find the `gazebo-assests-setup.sh` script.On that place I simply sourced the `assets-setup.sh` which doesnot work at all.

Hi @vikaskumareinstein! That's right, there is an error in the newest assets package and we are working on it. 

By now, you can install `jderobot-gazebo-assets` instead of `jderobot-assets`.

vikaskumareinstein, 2020-03-13:
I have just found out that after running `sudo apt install jderobot-gazebo-assets `  it creates the `gazebo-assets-setup.sh` script.After sourcing the script and  running `roslaunch ./launch/simple_line_follower_ros.launch` gazebo loads smoothly with required world and model.

zenithexpo, 2020-03-16:
Hello there, I'm working on Academy Challenge and facing the same issue of **"bash: /opt/ros/kinetic/setup.bash: No such file or directory"** while installation during "13. Export environment variables"
Can someone please point out the mistake I'm doing here or missed something..

diksha@diksha-HP-15-Notebook-PC:~/catkin_ws$ echo 'source '$PWD'/devel/setup.bash' >> ~/.bashrc
diksha@diksha-HP-15-Notebook-PC:~/catkin_ws$ echo 'export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/usr/share/gazebo-9' >> ~/.bashrc
diksha@diksha-HP-15-Notebook-PC:~/catkin_ws$ echo 'export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/opt/ros/melodic/share/jderobot_assets/models' >> ~/.bashrc
diksha@diksha-HP-15-Notebook-PC:~/catkin_ws$ source ~/.bashrc
bash: /opt/ros/kinetic/setup.bash: No such file or directory

Naman-ntc, 2020-03-16:
Hi, you are on the right track. Just switch from `/opt/ros/kinetic/setup.bash` to `/opt/ros/melodic/setup.bash` in your bash rc and you should be good to go

pariaspe, 2020-03-16:
Yes, that's it! Thank you @Naman-ntc

zenithexpo, 2020-03-16:
yes it worked, Thanks @Naman-ntc

NadineFadlallah, 2020-03-18:
Thank you all for this thread, I'm too working on the JdeRobot GSoC challenge (Wish me luck :D)
and I was intimated by the "bash: /opt/ros/kinetic/setup.bash: No such file or directory", and for a newbie it made no sense as I followed the instructions precisely (and it took  a huge amount of time!) 

for anyone encountering the same problem, what helped:
at step 13 & before (source ~/.bashrc)
cd ~
gedit .bashrc
find the line /opt/ros/kinetic/setup.bash substitute it with //opt/ros/melodic/setup.bash (Thanks @Naman-ntc)
cd catkin_ws

and continue as normal.
I hope they'll mention it in the instructions page soon :)

pariaspe, 2020-03-30:
New packages were released. Please make sure to update jderobot base and assets.

closed 2020-03-30 by pariaspe