JdeRobot/RoboticsAcademy #477: Color Filter Exercise

Naman-ntc, 2020-03-16:
Hi,
I wanted to ask if anybody is currently working on color filter exercise. While working for my "Academy Challenge", I built a working color-filter exercise with appropriate GUI for segmenting based on both RGB and HSV filters. I will soon provide my demo video too.
If you would like, I can try to provide my implementation/submit a PR.

Major updates made -
* Provide the required slider tools for filtering
* Provide a selection menu to select if required to filter using RGB or HSV or YUV color space
* In [color_filter.py](https://github.com/JdeRobot/RoboticsAcademy/blob/99642c1fb0d3d7bcc151babe22b57ae351ef3251/exercises/color_filter/color_filter.py#L86), threadCamera is set to run before the user presses play-code. This does not affect camera-based input but for local-videos, we skip over many frames till that time. So I call `t_cam.run()` from inside main window. 

I understand that `colorTuner` and `camServer` are still not ready and therefore you might not wish to work on color_filter till then. Let me know whatever you feel.

pankhurivanjani, 2020-03-16:
Hi @Naman-ntc !
I would like to comment on colortuner and camserver part.
1. What Is camserver? 
It just serves images..acts as a source...so you directly use usb_camera provided by ros 

2. Colortuner is ready. 
Binary package has been submitted and will be released soon. 
But, I would suggest you to use source installation if you are requiring it. Feel free to ask any query if you have.

Naman-ntc, 2020-03-16:
Thanks for the quick response @pankhurivanjani 
1) Oh, I see so it is an alternative to cv2.VideoCapture for webcam, and can also generalize to other camera devices such as external cameras!
2) I was using colorTuner installed with jderobots and came across an error, didn't know about the upcoming release. I don't have a hard requirement for colorTuner as it is not directly used in the exercise but thank you for letting me know.

jmplaza, 2020-03-16:
Hi @Naman-ntc ,

in order to be integrated some design principles have to be met:
1.-  Cameraserver will not be ready anymore, it has been replaced by ROS image drivers. Avoid direct image capture from the ROS node for this exercise. It should use a ROS driver (and so, ROS topics) to get the images, instead of getting the images directly through opencv. There are ROS drivers for webcams and for video files. The reference video files should be uploaded somewhere so any Academy student may check its results with them. 

2.- maybe we should change the focus for this exercise to make emphasis on it as an exercise, without sliders, just to show the effect of the student's code. No more than that, not another release of ColorTuner tool. The problem to be solved in this exercise would be: "locate the colored object in the image using a color filter". The student's code should apply a color filter to the image, estimate the object position in the image and a bounding box. The ROS template for this exercise should show in the GUI the filtered image, and also the estimated object position and the estimated bounding box.

ColorTuner may be used in the context of this exercise just to get the thresholds of the color filters. The student may check with ColorTuner the effect of several values for the color filter thresholds and then put those values on her code. But ColorTuner tool is not compulsory at all.

What do you think?

For each JdeRobot Academy exercise a webpage has to be created which includes the description of the problem, some useful theory and concepts to solve it, an illustrative video of the reference solution, an installation recipe (with links to the corresponding sections of Academy installation page), etc. The issue for the webpage of this exercise is #447

@jiban-jyoti has been working on this exercise too. He will submit soon a Pull-request with the his advances, so you can take his code into account too.

Naman-ntc, 2020-03-16:
Hi, @jmplaza 
1) I'll try to switch to ROS drivers. I like this approach as it would nice abstractions in the camera API switching between local-video and webcam using only a few commands.
2) Yes, this version of exercise makes more sense. Although I believe the sliders might still be useful since they would allow "finer" finetuning of colors which would affect the performance of the detector.  The user would write code for variable color ranges and apply the detection algorithm on top of it. It would probably allow a smoother flow.

Naman-ntc, 2020-03-20:
Hi,
So I was thinking about how to go around ros integrating with the exercise. Here is what I have collected
* For using the webcam (or any usb camera) `ros-usb_cam` drivers can be used. To use them, I would need to use the `cameraClient` API from jde-robots as done [in followface exercise](https://github.com/JdeRobot/RoboticsAcademy/blob/95404a2e40d07a4e0dbdfb8e2005ea80c73b4428/exercises/follow_face/gui/cameraSegment.py). 
* For local videos, I guess we would want, [video_stream_opencv](http://wiki.ros.org/video_stream_opencv) from ros. (By the way, I think video_stream_opencv also supports webcam or camera device apart from local videos so I think usb_cam is redundant right? We can use `video_stream_opencv` for both local videos and usb_cam)

Given that I have understood the above correctly, should I go ahead with implementation? I think I 
can finish and send a PR in 1-2 days. 
Here is what I plan to do --

1.) Create a launch file for local video/ camera device using video_stream_opencv driver (I can create different launch files too in case we want usb_cam). I will try to use cameraSegment from `followface` exercise
2.) Modify the [threadCamera](https://github.com/JdeRobot/RoboticsAcademy/blob/99642c1fb0d3d7bcc151babe22b57ae351ef3251/exercises/color_filter/color_filter.py#L86) cam.run() call on pressing the start button since :

> In color_filter.py, threadCamera is set to run before the user presses play-code. This does not affect camera-based input but for local-videos, we skip over many frames till that time. So I call t_cam.run() from inside the main window.

3.) Add the filter/slider GUI and object detection GUI as discussed above.

Naman-ntc, 2020-03-23:
Fixed by #481

closed 2020-03-23 by Naman-ntc