JdeRobot/RoboticsAcademy #485: Amazon Warehouse Exercise - Python Code error

sasilva1998, 2020-03-22:
Hi, I am trying the Amazon Warehouse exercise from [Amazon warehouse exercise](https://jderobot.github.io/RoboticsAcademy/exercises/MobileRobots/amazon_warehouse/). I was running this exercise about two days ago and was working well. Today I tried it again and I am getting the error below when running `python2 amazonWarehouse.py amazonMap.conf amazonConf.yml`. Today I upgraded some jderobot packages so I believe it could be with something like that. Can someone guide me trough this please?

```
Traceback (most recent call last):
  File "amazonWarehouse.py", line 25, in <module>
    import comm
  File "/opt/jderobot/lib/python2.7/comm/__init__.py", line , in <module>
    from .communicator import Communicator
  File "/opt/jderobot/lib/python2.7/comm/communicator.py", line 4, in <module>
    from .laserClient import getLaserClient
  File "/opt/jderobot/lib/python2.7/comm/laserClient.py", line 4, in <module>
    from .ice.laserIceClient import LaserIceClient
  File "/opt/jderobot/lib/python2.7/comm/ice/laserIceClient.py", line 21, in <module>
    import jderobot
  File "/opt/jderobot/lib/python2.7/jderobot/__init__.py", line 8, in <module>
    import ardroneextra_ice
  File "/opt/jderobot/lib/python2.7/ardroneextra_ice.py", line 164, in <module>
    _M_jderobot._t_ArDroneExtra = IcePy.defineClass('::jderobot::ArDroneExtra', ArDroneExtra, -1, (), True, False, None, (), ())
TypeError: function takes exactly 5 arguments (9 given)
```

jmplaza, 2020-03-22:
Hi @sasilva1998 , yes we have just release a new jderobot-base package which no longer use ICE as communication middleware. Instead it uses pure ROS communication and topics. This exercise has to be updated to work with only ROS. Sorry for the inconveniences....

sasilva1998, 2020-03-22:
Thanks for the response, though maybe I had damaged or erased some package :P

closed 2020-03-22 by sasilva1998

kurshakuz, 2020-03-24:
Hi, @sasilva1998. I was going to update this exercise and remove dependence on the comm package. I will keep you posted when everything will work again.

kurshakuz, 2020-03-24:
@sasilva1998 can you have a look again if the problem still there? I have built jderobot from scratch and cloned last RoboticsAcademy repository, and everything works fine.

sasilva1998, 2020-03-24:
@kurshakuz Hi, it is working fine now. I get the following output for `python2 -V`:
```
Python 2.7.17
```
Thanks for the help.