JdeRobot/RoboticsAcademy #493: Not doing source of ros melodic in the installation page

Klego, 2020-03-26:
Greetings.

After the installation of ros melodic, you should source the ros's setup.bash with this command:

echo 'source /opt/ros/melodic/setup.bash' >> ~/.bashrc
If you don't use this command, you can't use roslaunch.

kurshakuz, 2020-03-26:
Information about environmental setup is usually mentioned in the distro documentation itself. For example, in [1] point 1.5 already mentions that.

[1] http://wiki.ros.org/melodic/Installation/Ubuntu

Klego, 2020-03-26:
Yeah, but the students are told to follow the instructions and most of them don't have any idea of ROS and how it works. 

So if the instructions are incomplete... it's an issue.

nkhedekar, 2020-03-26:
In this case, I'd agree with @Klego as the page does not provide a link to the official installation. In fact the instructions currently present, miss a few steps for setting up rosdep and installing other essentials that are mentioned in the official installation. These should also be added and I think it would be nice to put in the link for official instructions as well.

nkhedekar, 2020-03-27:
Solved in #495.

Thanks @klego @dcharrezt, @ojitmehta123

closed 2020-03-27 by nkhedekar