JdeRobot/RoboticsAcademy #494: models not loading in follow_line exercise

itsyashkothari, 2020-03-26:
Ubuntu 18.04
Ros: Melodic 
Installed jderobot and jderobot-assets.
Sourced ros,gazebo-setup and asset-setup. 

When i try to run ./launch/simple_line_follower_ros.launch, gazebo opens but no model is visible.
[image]
Did i do something wrong?

ojitmehta123, 2020-03-26:
Hi @itsyashkothari. I had faced similar issues. There were some bugs in the previous release of jderobot-assets which have been rectified and the new release will be out soon.
It worked for me by editing the export of GAZEBO_RESOURCE_PATH and GAZEBO_MODEL_PATH in file /opt/jderobot/share/jderobot/gazebo/assets-setup.sh as:  
```bash
export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/opt/jderobot/share/jderobot/gazebo/models:/opt/jderobot/share/jderobot/gazebo/worlds
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/opt/jderobot/share/jderobot/gazebo/models
```  
I hope this solves your problem as well.  
Thanks

itsyashkothari, 2020-03-26:
Hey @ojitmehta123, thanks a lot. Worked like a charm. 
I am planning to apply for gsoc, is there any IRC for that?

jmplaza, 2020-03-26:
Hi @itsyashkothari , no there is not any IRC but you have the [organization mailing list](https://jderobot.github.io/community/) and the [candidate selection process instructions](https://jderobot.github.io/activities/gsoc/2020#application-instructions-for-gsoc-2020). Hurry up, you are a bit late...

itsyashkothari, 2020-03-26:
Thanks for the reply. Will do it ASAP.

closed 2020-03-26 by itsyashkothari