JdeRobot/RoboticsAcademy #528: [Github pages] OpenCV version in Installation instructions

diegomrt, 2020-04-20:
It seems that the last OpenCV version (4.2.0) is causing segmentation faults when launching the exercises RQT-based GUI, probably due to some incompatibilities between QT libraries and OpenCV 4.2.0

Installation instructions must be updated to account for this. A good idea could be to suggest installation via `sudo apt-get install libopencv-dev python-opencv`, which at the moment installs version 3.2.0.

Sangu80, 2021-03-24:
@diegomrt this might help you out
ep 1 – As we are going to build our own OpenCV on the system need to install all required dependencies on the system, you can use the following command to install OpenCV dependencies:

$ sudo apt install build-essential cmake git pkg-config libgtk-3-dev \
    libavcodec-dev libavformat-dev libswscale-dev libv4l-dev \
    libxvidcore-dev libx264-dev libjpeg-dev libpng-dev libtiff-dev \
    gfortran openexr libatlas-base-dev python3-dev python3-numpy \
    libtbb2 libtbb-dev libdc1394-22-dev
Step 2 – Now make a clone of OpenCV’s and OpenCV contrib repository from git to your system.

$ mkdir ~/opencv_build && cd ~/opencv_build
$ git clone https://github.com/opencv/opencv.git
$ git clone https://github.com/opencv/opencv_contrib.git
Step 3 – After completing the download of the package, create a build directory and enter into the directory, as shown below:

$ cd ~/opencv_build/opencv
$ mkdir build && cd build
You can set up the OpenCV build using the CMake:

$ cmake -D CMAKE_BUILD_TYPE=RELEASE \
    -D CMAKE_INSTALL_PREFIX=/usr/local \
    -D INSTALL_C_EXAMPLES=ON \
    -D INSTALL_PYTHON_EXAMPLES=ON \
    -D OPENCV_GENERATE_PKGCONFIG=ON \
    -D OPENCV_EXTRA_MODULES_PATH=~/opencv_build/opencv_contrib/modules \
    -D BUILD_EXAMPLES=ON ..
Step 4 – Now start compilation process for the latest build using makes the command, as shown below:

$ make -j8
Step 5 – Now, after compile, you can install OpenCV only using the make command, as shown below:

$ sudo make install
Step 6 – You can verify the OpenCV successfully installed in your system by checking the version of installed OpenCV:

$ pkg-config --modversion opencv4

ReyDoran, 2021-05-07:
The recommended way to run Robotics Academy exercises is now using the RADI image. This image already installs opencv version 3.2.0 using such command (`sudo apt-get install libopencv-dev`)

closed 2021-05-07 by ReyDoran