JdeRobot/RoboticsAcademy #529: Hot-reloading student code

diegomrt, 2020-04-20:
It would be a great enhancement for the Academy if once the student updates his/her code (at solution.py), this updated solution is executed at the exercise GUI when pressing "Play code" button, without relaunching the exercise.

jmplaza, 2020-04-20:
@fqez has developed something similar for BehaviorSuite. We could reuse it here

jmplaza, 2020-11-26:
We are working on this. The hot reloading of student's code is a feature in the new release of RoboticsAcademy, 2.3. It is provided by the design of the Web template for each exercise. Currently only two exercises are in 2.3 release, the [FollowLine](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/follow_line/#instructions-for-web-templates) and the [ObstacleAvoidance](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/obstacle_avoidance#instructions-for-web-templates). As long more exercises are migrated to 2.3 release all of them will eventually have this hot-reloading feature.

closed 2020-11-26 by jmplaza