JdeRobot/RoboticsAcademy #556: PX4 Error while setting up Drone Gymkhana

joeljosephjin, 2020-06-25:
Command:
```
$ roslaunch drone_gymkhana.launch
```

Specific Error:
```
ERROR: cannot launch node of type [px4/px4]: Cannot locate node of type [px4] in package [px4]. Make sure file exists in package path and permission is set to executable (chmod +x)
```

Whole Output:
```
... logging to /home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/roslaunch-elian-ThinkPad-L440-23850.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://elian-ThinkPad-L440:44675/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
[... 227 lines cut ...]
  File "/home/elian/Documents/drone_gymkhana/my_solution.py", line 8, in <module>
    from drone_wrapper import DroneWrapper
  File "/opt/ros/melodic/lib/python2.7/dist-packages/drone_wrapper/__init__.py", line 1, in <module>
    from drone_wrapper_class import DroneWrapper
ModuleNotFoundError: No module named 'drone_wrapper_class'
[my_solution-7] process has died [pid 23879, exit code 1, cmd /opt/ros/melodic/lib/drone_wrapper/play_python_code /home/elian/Documents/drone_gymkhana/my_solution.py __name:=my_solution __log:=/home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/my_solution-7.log].
log file: /home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/my_solution-7*.log
[INFO] [1593070151.211301, 0.000000]: Loading model XML from file /opt/ros/melodic/share/jderobot_assets/models/iris_dual_cam/iris_dual_cam.sdf
[INFO] [1593070151.254365, 499.564000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1593070151.257004, 499.568000]: Calling service /gazebo/spawn_sdf_model
[ERROR] [1593070151.403532138, 499.716000000]: SpawnModel: Failure - model name iris already exist.
[INFO] [1593070151.447905, 499.732000]: Spawn status: SpawnModel: Failure - entity already exists.
[ERROR] [1593070151.450624, 499.732000]: Spawn service failed. Exiting.
[vehicle_spawn_elian_ThinkPad_L440_23850_2455365306996860541-4] process has died [pid 23874, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /opt/ros/melodic/share/jderobot_assets/models/iris_dual_cam/iris_dual_cam.sdf -model iris -x 0 -y 0 -z 1 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_elian_ThinkPad_L440_23850_2455365306996860541 __log:=/home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/vehicle_spawn_elian_ThinkPad_L440_23850_2455365306996860541-4.log].
log file: /home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/vehicle_spawn_elian_ThinkPad_L440_23850_2455365306996860541-4*.log
[ INFO] [1593070154.595575629]: Finished loading Gazebo ROS API Plugin.
[ WARN] [1593070154.597550140, 502.572000000]: Shutdown request received.
[ WARN] [1593070154.597828063, 502.572000000]: Reason given for shutdown: [new node registered with same name]
[ INFO] [1593070154.616133833]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[gazebo-2] process has died [pid 23866, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/elian/Documents/drone_gymkhana/drone_gymkhana.world __name:=gazebo __log:=/home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/gazebo-2.log].
log file: /home/elian/.ros/log/55013450-b6b4-11ea-8e42-e8b1fc0f6757/gazebo-2*.log
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
```

pariaspe, 2020-06-25:
Hi @joeljosephjin,

We made some changes that will be available in the next ROS sync (in 10d more o less). However, if you want to try it right now you can clone [Assets](https://github.com/JdeRobot/assets) and [Drones](https://github.com/JdeRobot/drones) repositories into your catkin_ws/src and rebuild it.

diegomrt, 2020-06-26:
Hi @joeljosephjin 

In addition, it seems that you have some troubles with the PX4 (Firmware) package. MavROS is launching OK in your system but the PX4 SITL fails. Did you get any errors when compiling your catkin_ws?

To check it, try to run `roslaunch px4 posix_sitl.launch` and let us know if it works.

pariaspe, 2021-04-04:
ROS didn't find PX4 package, might have some troubles with `ROS_PACKAGE_PATH` envvar. Try `roslaunch px4 mavros_posix_sitl.launch` to confirm.

Stale issue. Closing.

closed 2021-04-04 by pariaspe