JdeRobot/RoboticsAcademy #561: Error while transforming filtered image to grayscale

NayOoLwin5, 2020-07-28:
In follow line exercise,when transforming the filtered image to grayscale for finding the center point of the red line, it shows tuple index out of range error indicating the cameraWidget.py file. Anything to change in that file for grayscale?
```
Traceback (most recent call last):
  File "/home/dell/Academy/exercises/follow_line/gui/GUI.py", line 37, in updateGUI
    self.camera1.updateImage()
  File "/home/dell/Academy/exercises/follow_line/gui/widgets/cameraWidget.py", line 31, in updateImage
    image = QtGui.QImage(resized.data, resized.shape[1], resized.shape[0], resized.shape[1]*resized.shape[2], QtGui.QImage.Format_RGB888)
IndexError: tuple index out of range
Aborted (core dumped)

```
Tried changing RGB888 to Grayscale8. Still the same.

SakshayMahna, 2020-07-29:
Hello!
Are you trying to display a grayscale image in the GUI template?
Can you post a code snippet of how your algorithm is using the API for this?

NayOoLwin5, 2020-07-30:
Hi Sakshay,
Yes I was trying to display it on GUI template.Now I could display.I was defining that grayscale in different function which is why getting error.
Thanks for your response.

closed 2020-07-30 by NayOoLwin5