JdeRobot/RoboticsAcademy #576: error while running catkin build

abubakrsiddq, 2020-08-06:
[image]

i followed all the steps listed,but im stuck her.
Could you tell me how to resolve this issue

SakshayMahna, 2020-08-10:
Hello!
From the image it looks like you are trying to run the specific infrastructure installation instructions.

What is the exercise that you are supposed to run?

abubakrsiddq, 2020-08-10:
i was following instructions given here
https://jderobot.github.io/RoboticsAcademy/installation/

jmplaza, 2021-03-29:
Please jump to RoboticsAcademy 2.3. It is web-based and all the dependencies are pre-installed in the RADI (=RoboticsAcademy Docker Image). 2.0 and 2.1 releases are no longer supported.

closed 2021-03-29 by jmplaza