JdeRobot/RoboticsAcademy #592: Running Turtlebot2 on ROS Melodic with Ubuntu 18.04

jmvega, 2020-09-06:
Turtlebot2 is no longer supported on ROS Melodic.

jmplaza, 2020-09-19:
Hi @jmvega could you please record a demo video about it? Is this issue closeable then?

fqez, 2020-11-13:
Hi @jmvega I managed to make it work long time ago with ROS Melodic and ubuntu 18.04. I made a script that install the necessary pieces to make it work. Check it out, maybe you find it useful in some way :)

https://github.com/JdeRobot/base/tree/master/src/drivers/turtlebot_ws

trunc8, 2021-04-02:
Hi @jmplaza, is this issue still open?

jmplaza, 2021-04-02:
Hi @trunc8 , do you have access to a real Turtlebot2?

trunc8, 2021-04-02:
Sorry, I do not have access to physical Turtlebot2 hardware, but I have worked with Turtlebot2 in Kinetic before. I can perform any software tests if that can be of any help!

jmplaza, 2021-04-03:
Hi, 
no, this issue is just for tests with the real robot...

jmplaza, 2021-11-21:
AFAIK @Carlosalpha1 you are already working on this, not for ROS-Noetic, but for ROS2-Foxy. Aren't you?

Carlosalpha1, 2021-11-21:
Yes, @jmplaza, in a first approximation, I have the integration of kobuki_keyop for ROS2-Foxy [**here**](https://hub.docker.com/r/carlosalpha1/kobuki_keyop)