JdeRobot/RoboticsAcademy #616: [Web Templates] Automate the Docker Launch

SakshayMahna, 2020-10-13:
Currently, there are a lot of instructions to launch the Docker Container to run a particular exercise. 3 separate Docker terminals required to run a single exercise.

Need to add a `manager.py` file to automate this process.

NayOoLwin5, 2020-10-14:
Hi @SakshayMahna,
            
Let me handle that part. Just want to make sure that `RoboticsAcademy/scripts/manager.py` is the main docker file to be modified, isn't it?
Or any different `Dockerfile` which handles the image?

SakshayMahna, 2020-10-15:
Hello!
Actually, no changes are required for this file as such. Copy pasting from the branch `issue-596` is to be done, and some Javascript files to enable communication through Websockets.

Thanks for offering the help, but I am almost done with this. Just some errors left to be debugged.

The current commit on the branch `issue-616` contains the current work for the Follow Line exercise. You may try and complete it for Obstacle Avoidance one.

closed 2020-10-17 by SakshayMahna