JdeRobot/RoboticsAcademy #625: web browser shows connection closed in follow-line exercise.

divgup, 2020-10-23:
This is terminal screenshot
[image]

Web browser shows on opening index.html
[image]

Also I tried implementing the instructions that were there 5-6 days ago on the website page to do the follow line exercise with 
web templates. But no simulation window appears as shown in attached video at the exercise page. However connection gets established but no Gazebo simulation  window opened. 

How to resolve the issue?
Thanks in advance

NayOoLwin5, 2020-10-23:
Hi, assuming your host machine is linux..
Check the websocket_address variable inside `web-template/assets/websocket_address.js` on your local machine. If it is **127.0.0.1** localhost then it  should work fine.

If the problem still persists, try changing to this address **127.0.0.53** which is the address of the local caching stub resolver and through which the docker can communicate too.
Now, the alert should show connection established. Let us know if it works.

divgup, 2020-10-28:
Hi @NayOoLwin5 
Now both the address(127.0.0.1 and 127.0.0.53) are working perfectly fine. I don't know what was the problem before.
Thanks

closed 2020-10-28 by divgup