JdeRobot/RoboticsAcademy #645: Error while installing jderobot `sudo apt-get install jderobot`

vikaskumareinstein, 2020-11-18:
I am getting this error when i tried installing generic infrastructure for RoboticsAcademy. Every step worked fine before this step:
**
Install JdeRobot-base

 sudo apt install jderobot
**
I am getting this error:
`Err:1 http://wiki.jderobot.org/apt bionic/main amd64 jderobot-config amd64 6.1.1
  502  Proxy Error [IP: 193.147.79.60 80]
E: Failed to fetch http://wiki.jderobot.org/apt/pool/main/j/jderobot-config/jderobot-config_6.1.1_amd64.deb  502  Proxy Error [IP: 193.147.79.60 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?`

I also tried installing it offline from this ` http://wiki.jderobot.org/apt/pool/main/j/jderobot-config/jderobot-config_6.1.1_amd64.deb` site but it doesnt download. The jderobot-config_5.6.8_amd64.deb gets downloaded but not this one.

Please help! and thanks in advance.

vikaskumareinstein, 2020-11-18:
a small fix:
I just installed the version of `jderobot-config_5.6.8_amd64.deb`  from `http://wiki.jderobot.org/apt/pool/main/j/jderobot-config/jderobot-config_5.6.8_amd64.deb`and then it is working fine after running `sudo apt-get install jderobot`.
But still the latest `jderobot-config_6.1.1_amd64.deb` is unable to download. Kindly look into it.

jmplaza, 2021-03-01:
Jderobot-base or jderobot-assets are no longer a requirement for exercises in RoboticsAcademy 2.1 and 2.3

closed 2021-03-01 by jmplaza