JdeRobot/RoboticsAcademy #650: Enable USB/serial communication with real robots and other hardware from the docker image

rocapal, 2020-11-26:
Enable communications (USB, serial, etc) from the docker image to allow interaction with hardware physically connected to the host. Robots, cameras, sensors and others must be accessible from the docker image.

Use the docker image of robotics-academy [1] for testing.

[1] https://hub.docker.com/r/jderobot/robotics-academy/tags

rocapal, 2020-11-26:
Summary: It is possible to map devices from the host to the docker image. There are several ways to do that depending on our needs. They are described below. 

## Connect a single device

Having the following image in the system
```
$ docker images
REPOSITORY                  TAG                 IMAGE ID            CREATED             SIZE
jderobot/robotics-academy   latest              dc94eb225ca6        8 days ago          3.76GB
```
We can run the image using the **--device** parameters to tell docker to map that device inside the docker.
```
$ docker run -t -i --device=/dev/video3 dc94eb225ca6 bash
```
In this case, /dev/video3 is an USB camera connected to the host. And we can access to it from the docker image:
```
root@40e24524e252:/# v4l2-ctl --all -d /dev/video3
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : USB Camera: USB Camera
	Bus info      : usb-0000:00:14.0-1.2
	Driver version: 5.9.9
	Capabilities  : 0x84A00001

```
Test it with webcams, usb devices and arduino boards. All of them work fine. However, this method does not allow the hot plug/unplug of the devices (see next option).

## Connect all devices and allow hot plug/unplug  

This method might not be a secure option for obvious reasons depending on the scenario/circumstances. But, it allows to map the full USB bus into the docker and detect the hot plug/unplug of the devices without restarting the docker.

```
$ docker run -t -i --privileged -v /dev/bus/usb:/dev/bus/usb dc94eb225ca6 bash
```
Notice that using '-v' option we can pass any type of volume (even a single device). More info in the following links:

https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities
https://docs.docker.com/storage/volumes/

gbernalr, 2023-08-03:
This problem of connectivity between usb inside the docker is solved in this github pages entry where the physical drivers of the turtlebot2 are used, where the camera, the engines and the rplidar are linked to be used inside the docker.
In general, it is enough to know which usb-driver is linked to each physical driver and set it when running the docker.
For example with:
--device /dev/ttyUSB0
With this flag we establish that inside the docker we will use that usb.
https://roboticslaburjc.github.io/2022-tfg-guillermo-bernal/weblog/Semana_7/