JdeRobot/RoboticsAcademy #664: Remove simulator sidebar Gzweb  RADI  (2.3 WebTemplates & Unibotics)

martaquintana, 2020-12-11:
Eliminar barra lateral del simulador Gzweb en el RADI ( Para 2.3 WebTemplates & Unibotics)

PASOS:
Entrando en gzweb/http/client/gz3d.gui.js. Busca el siguiente codigo (línea 64946 para gzweb 1.4):

if (isWideScreen())
   {
     globalEmitter.emit('openTab', 'mainMenu', 'mainMenu');
   }
   ...

añadir:

globalEmitter.emit('closeTabs', true);

quedaría así:

if (isWideScreen())
    {
        globalEmitter.emit('openTab', 'mainMenu', 'mainMenu');
        globalEmitter.emit('closeTabs', true);
    }
    ...

de esta manera ya estaría oculta la barra lateral.

Suhas-G, 2021-02-28:
Hello. I'm a new user of RoboticsAcademy. I was going through the Issues list, and this seemed like a nice first issue (because you have already provided a solution :smile: ). But 'gzweb/http/client/gz3d.gui.js' appears to be a minified file under [gzweb](https://github.com/osrf/gzweb) repository. So any change will have to be a pull-request on that...

I tried to find whether its possible to inject code into an iframe once its loaded. But this method is restricted by browser when the domains are different. Since 'locahost:8080' is used for gzweb and exercise can be opened just as a file, I was not able to proceed in this way.

Am I missing something? Any suggestions?

jmplaza, 2021-04-04:
As we are replacing the Gazebo scenario viewer in all exercises from gzweb to novnc this issue has been superseded by https://github.com/JdeRobot/RoboticsAcademy/issues/790

closed 2021-04-04 by jmplaza