JdeRobot/RoboticsAcademy #683: [Documentation] Update Web Page for Color Filter Exercise

jessiffmm, 2020-12-30:
Update the Web Page Documentation of the Color Filter Exercise.

Ajayjeena, 2021-01-30:
How can i do the same can you please help me as i am new here

jessiffmm, 2021-01-30:
Do you ask about how to do the documentation or how to create the issue?

Ajayjeena, 2021-01-30:
Yes I'm interested in computer vision and robotics. So I wanna contribute
towards the same and learn latest technologies.

jessiffmm, 2021-01-30:
Do you participate in jderobot?

Ajayjeena, 2021-01-30:
No not till now but wanna participate in it.

jessiffmm, 2021-01-30:
ok, you should talk to jmplaza. It is the person who leads jderobot.

Ajayjeena, 2021-01-30:
Okk thank you so much for giving me your valuable time.

jessiffmm, 2021-01-30:
You're welcome! If you need anything else do not hesitate to ask!

Ajayjeena, 2021-01-30:
Thank you so much . I will ask

closed 2021-03-12 by dvalladaresv