JdeRobot/RoboticsAcademy #686: bash: /opt/ros/noetic/setup.bash: No such file or directory

snehal1720, 2021-01-02:
[image]

bash: /opt/ros/noetic/setup.bash: No such file or directory. How to fix this??

pawanw17, 2021-01-03:
Did you install ros noetic correctly?

jmplaza, 2021-01-03:
Hi @snehal1720, 

each exercise requires a set of dependences, as stated in each exercise webpage and in the [general installation webpage](
https://jderobot.github.io/RoboticsAcademy/installation). Currently ROS1-Melodic is the reference distribution for RoboticsAcademy and the only one supported, with a working installation recipe for the dependences of each exercise. There are plans to migration to ROS1-Noetic and even ROS2-Foxy, but they are only prototypes so far. Volunteer developers are welcome there!

In addition, feel free to use the RADI ([RoboticsAcademy Docker Image](https://hub.docker.com/r/jderobot/robotics-academy/tags)) where many dependencies are already pre-installed and ready to use. You may use current RoboticsAcademy from that docker image.

pawanw17, 2021-01-03:
Hey @jmplaza I can help in the migration, can you create a branch for noetic to submit any future PRs?

jmplaza, 2021-01-03:
Hi @pawanw17, 

we are in the process of migrating all exercises from current RoboticsAcademy 2.0 or 2.1 (=using ROS nodes in Python as exercise templates) towards 2.3 (=using web based exercise templates). RoboticsAcademy 2.3 uses a Docker Image with all the dependences already preinstalled, currently that RADI has ROS-Melodic inside. All users may download the latest RADI from [dockerhub](https://hub.docker.com/r/jderobot/robotics-academy/tags). Not all the exercises are already migrated to 2.3, just  FollowLine, ObstacleAvoidance, VacuumCleaner and LocalizedVacuumCleaner. 

The best way of contributing is to help in that migration: (a) to get familiar and do betatesting on those four exercises at 2.3, providing feedback here or in the [RoboticsAcademy forum](https://forum.unibotics.org/); (b) migrating new exercises to 2.3 is also a good way of contributing. 

Once we got all the exercises available in 2.3 then we will work on updating RADI from ROS-Melodic to ROS-Noetic.

snehal1720, 2021-01-04:
I am trying to install ros melodic but I am encountering an error. Kindly see the attached screenshot and help me at the earliest.
[image]

snehal1720, 2021-01-04:
I tried solving this error but I am not able to resolve it. Kindly help.

jmplaza, 2021-01-04:
Hi @snehal1720 ,

instead of running Robotics-Academy 2.1 (which requires local installation of all the dependences, incluiding ROS-Melodic) you better run 2.3. Check the [new installation instructions](https://jderobot.github.io/RoboticsAcademy/installation/), we have just updated them for 2.3 users.

pawanw17, 2021-01-04:
Thanks @jmplaza for the insights! will start by beta testing the excercises.

jmplaza, 2021-03-29:
ROS1 Melodic is the only ROS distribution currently supported. Anyway the users are not required to install it on their computers at all, as it is already preinstalled in the RADI (RoboticsAcademy Docker Image). 

The plan is to shift to ROS1 Noetic inside RADI (and even ROS2 Foxy) after migrating most exercises to RoboticsAcademy 2.3. New RADIs will be created for them.

closed 2021-03-29 by jmplaza