JdeRobot/RoboticsAcademy #737: Optimization of Gazebo execution inside the RoboticsAcademy Docker Image (RADI)

jmplaza, 2021-02-17:
In case of nvidia graphics cards and Linux hosts, there is a chance of mapping the nvidia driver (and so graphics acceleration with GPU) in the host machine from the docker image itself. With that mapping the Gazebo execution in the docker image would be much faster than with only CPU.

Current Gazebo execution from RADI does not take benefit of the GPU of the host machine. It always uses a xvfb XWindow server inside the RADI, and so, no graphics acceleration is used at all.

SakshayMahna, 2021-02-18:
Xvfb update to Xdummy and advanced instructions to enable GPU acceleration have been merged in PR #734 

Testing yet to be done

SakshayMahna, 2021-03-17:
One optimization on Browser side: replaced `<canvas>` tag with `<img>` tag.

jmplaza, 2021-05-06:
https://github.com/turlucode/ros-docker-gui seems to be a nice and recent proof of concept. He includes ros-melodic and it departs from nvidia/opengl:1.0-glvnd-runtime-ubuntu18.04 as base docker image

ReyDoran, 2021-05-18:
There is a new branch on this repository [RADI-2.4.0-beta](https://github.com/JdeRobot/RoboticsAcademy/tree/RADI-2.4.0-beta) based on [Jovian-Dsouza fork](https://github.com/Jovian-Dsouza/RoboticsAcademy/tree/nvidia_radi) which already has some performance improvements. It is not ready to be merged with master as it has some error displaying GUI images and is not tested enough on different machines and OS. There are currently 2 issues: [issue-958](https://github.com/JdeRobot/RoboticsAcademy/issues/958) and [issue-957](https://github.com/JdeRobot/RoboticsAcademy/issues/957)
The built image is uploaded to dockerhub with [tag 2.4.0-beta](https://hub.docker.com/layers/jderobot/robotics-academy/2.4.0-beta/images/sha256-0ec58e38cc1d1a46ffc1550ddeb7a20d96f606447c49573015c6809af5ffc4ed?context=explore).

jmplaza, 2021-05-18:
@lr-morales @fqez this RADI is the one it would be great you test

ReyDoran, 2021-06-07:
RADI 2.4.0 is now ready and supports hardware acceleration for linux + nvidia users. It also supports the other scenarios without acceleration. Currently it is only tested in depth on Follow Line exercise. I will create a new issue to expand it to the remaining exercises. As stated on the official RoboticsAcademy channel we are going to have two Dockerfiles and two images (one with :2.4 suffix) to progressively migrate each exercise.

ReyDoran, 2021-06-07:
This issue is now closed by the following commits:
- Documentation [abd68577e99f446cff155d60c8ea450b329f778b](https://github.com/JdeRobot/RoboticsAcademy/commit/abd68577e99f446cff155d60c8ea450b329f778b)
- [RADI 2.4.0 merge](https://github.com/JdeRobot/RoboticsAcademy/pull/985)

[New issue](https://github.com/JdeRobot/RoboticsAcademy/issues/997) to expand RADI 2.4.0 to the remaining exercises

closed 2021-06-07 by ReyDoran