JdeRobot/RoboticsAcademy #741: Connection closed after syntax error in exercise Follow Line

Suhas-G, 2021-02-24:
The issue is that if there is any syntax error in python code in browser, after pressing Play, the UI becomes unusable and a 'Connection closed' message is shown. So I end up restarting the container. Its not a blocker in getting the exercise to work, but please do suggest if there's a way to avoid 'Connection closed' state.

Steps to reproduce: 
1. From https://hub.docker.com/r/jderobot/robotics-academy the command `docker build -f image-name .` didn't work, so I used `docker build -t local_robotics_academy:latest  .`
2. Ran `docker run -it --name=jderobots -p 8080:8080 -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 local_robotics_academy python3.8 manager.py`
3. Open the 'exercise.html' under RoboticsAcademy/exercises/follow_line/web-template in browser.
4. Write some code which has syntax error and click on 'Play'
5. Now, both the 'Play' and 'Stop' button cannot be used anymore even if the syntax error is corrected. After sometime, a message 'Connection closed' is automatically shown.

Before sending erroneous code. (There is a typo 'cv' instead of 'cv2' on line 18)
[image]

After pressing Play for erroneous code and trying to press either Play or Stop after correcting the code.
[image]

Error present in docker bash 
[image]

SakshayMahna, 2021-02-24:
Hello!

Thank You for pointing that out! Some points regarding your observations:

- For the documentation, the command should be `-t` instead of `-f`
- Regarding the connection closed, this is a very specific error that you observed. Generally, the errors that are in the syntax(Python code of User) are displayed in the Pseudo-Console(activated by clicking the terminal like looking button to the right of Gazebo button).

However, this error seems to be more related to the Javascript part, rather the Docker image part. I will get back after analyzing this issue.

Suhas-G, 2021-02-24:
I investigated the JS part a bit where it was throwing the exception, so in case of a Python error, the `data.text_buffer` is not in JSON format. So I added a try catch block as a quick fix
```javascript
try {
        messages = JSON.parse(data.text_buffer);
} catch (error) {
	messages = [data.text_buffer];
}
``` 
And I was able to 'Stop' or 'Play' even after trying out erroneous code. Same way, I also tried adding the below lines before JSON parse and that also worked, even though the parsing error still happens. 
```javascript
// Send the Acknowledgment Message
websocket_gui.send("#ack"); 
```
So I'm guessing the python server is timing out waiting for acknowledgement. (I haven't checked this part of code yet)

SakshayMahna, 2021-02-26:
Hello!

I have found where the exact problem is(after doing some `console.logs`). The Javascript part that you referred above, receives a list of strings from the Python `host.py` running inside RADI. The `JSON.parse` is supposed to interpret that from a string format to a list format.

`host.py` on line 200, has a line of code `self.console.print(exec_value)`. `exec_value` is not a string, it should be converted to a string before sending it through the websocket. Since, it is not a string, `JSON.parse()` function is not able to interpret it, hence giving the error.

In order to solve the whole problem, a change should be made in the  `print` function of `console.py`, to always type convert what it receives to a string before appending it to it's buffer.

SakshayMahna, 2021-02-26:
Since, this is a problem for all the 4 current exercises. I'll submit a PR for the same correcting this in all the 4 exercises along with the documentation error.

A similar problem is faced by a user at the [Unibotics Forum](https://forum.unibotics.org/t/errors-doesnt-apears-anywhare/306).

Suhas-G, 2021-02-26:
Great, thanks!

closed 2021-02-26 by pankhurivanjani