JdeRobot/RoboticsAcademy #746: Error handling for ws_code.js  [Follow line exercise]

shreyas2000, 2021-02-26:
There are two issues when we launch exercise.html for [Follow line]:
1)If we click on the play button before "_connection established_" pop up comes , we get a error in console log as seen in the below snapshot. This is because play button calls the **submitcode()** function in **ws_code.js** which executes **websocket_code.send()** function. But the value of variable "_python_code_" comes after connection is established.

2)The console log from the above **submitcode()** function "Code Sent! Check terminal for more information!"  gets logged even without  **websocket_code.send()** function  running  i.e every time you click the play button you get the console log.

[image]

What i thought we can do is simply add try catch like so :
```
try{
var python_code = editor.getValue();
    python_code = "#code\n" + python_code
    
    // Get the debug level and add header
	var debug_level = document.querySelector('input[name = "debug"]').value;
    python_code = "#dbug" + debug_level + python_code
    websocket_code.send(python_code);
    console.log("Code Sent! Check terminal for more information!");
    
	}
	catch{
		alert("Please wait till connection gets established");
	}
```
from line number 58 in ws_code.js to solve first error
and put console.log below   `websocket_code.send(python_code);` / use .then for second issue.

SakshayMahna, 2021-02-27:
Hello!

Although, it is mentioned in the Documentation, that we have to wait for Connection Established message. 

But having this type of exception catching would be good, in case a user sends code before connecting.

shreyas2000, 2021-02-27:
Hi!
Yes it will be good if we took care of that. Please close the issue after you are done making necessary changes.

closed 2021-05-10 by ReyDoran