JdeRobot/RoboticsAcademy #756: Create a home page of exercises

dvalladaresv, 2021-03-09:


shreyas2000, 2021-03-11:
How should it be different from the exsisting [exercises](http://jderobot.github.io/RoboticsAcademy/exercises/) page. What information needs to be included in it. Do you intend to use it instead of exsisting [exercise](http://jderobot.github.io/RoboticsAcademy/exercises/) page

dvalladaresv, 2021-03-11:
This issue is to create an html template that allows you to enter the exercises directly. That is to say that you can access from it to the existing exercises.html. 

It is not related to change the web page of the exercises documentation.

shreyas2000, 2021-03-11:
ok got it!

shreyas2000, 2021-03-11:
So user runs the docker image on his machine and acesses the exercise from home page of exercise.

dvalladaresv, 2021-03-12:
Yes, that's the idea

PR #776 (2021-03-20): exercises main page
Roboticsacademy.html has been created in the root of the project, which contains a grid with the accesses to the exercises with the available web template

Fix #756
closed 2021-03-20 by dvalladaresv