JdeRobot/RoboticsAcademy #781: error when launching exercise 'drone_cat_mouse'

hibetterheyj, 2021-03-24:
Dear developers,

I have encountered an error when launching the web template of 'drone_cat_mouse' as follows:
```shell
~$ docker run -it -p 8080:8080 -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 jderobot/robotics-academy:drones-beta python3.8 manager.py
['/', '/opt/ros/melodic/lib/python2.7/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']

<websockets.server.WebSocketServerProtocol object at 0x7fb08df4ee80>
{"command":"exit","exercise":""}
ALL KILED
{"command":"open","exercise":"drone_cat_mouse"}
> Starting simulation
Error in connection handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/websockets/server.py", line 191, in handler
    await self.ws_handler(self, path)
  File "manager.py", line 96, in hello
    roslaunch_cmd = ros_instructions(data["exercise"])
  File "manager.py", line 47, in ros_instructions
    roslaunch_cmd = roslaunch_cmd + export_gazebo(exercise)
  File "manager.py", line 39, in export_gazebo
    gazebo_path = GAZEBO_RESOURCE_PATH + instructions[exercise]["gazebo_path"] + ";"
KeyError: 'drone_cat_mouse'
```

Therefore, I would like to seek suggestions from you. Thanks in advance~

pariaspe, 2021-03-25:
Hi @hibetterheyj 
Please try to launch the exercise with the latest RADI. Drones-beta version might be deprecated.

```bash
docker run -it -p 8080:8080 -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 jderobot/robotics-academy python3.8 manager.py
```

Notice that you can also launch the exercise running the script `RADI-launch`.

hibetterheyj, 2021-03-25:
Thanks for your reply!

jmplaza, 2021-03-26:
Hi @hibetterheyj , feel free to close this issue if using the latest RADI and the right launching instructions for this exercise work fine for you.

closed 2021-03-26 by hibetterheyj