JdeRobot/RoboticsAcademy #783: Improve the Gazebo scenario in FollowLine exercise

jmplaza, 2021-03-25:
to include the sky with clouds, instead of the bland grey "sky". It may require some additions in the [CustomRobots repository](https://github.com/JdeRobot/CustomRobots), as it contains the assets for this exercise.

Spnetic-5, 2021-03-28:
Can you assign this issue to me, I'm new to open source and really interested in contributing to JdeRobot.

jmplaza, 2021-03-28:
There is no need of an explicit assignment, just work on it. The sooner you send your Pull-Request solving this issue the higher the chances of being merged. Cheers!

Spnetic-5, 2021-03-29:
Ohh Nice, can you please tell me how my approach should be?

jmplaza, 2021-03-29:
Hi @Spnetic-5 , 
before contributing it would be useful you learn enough background on ROS robotics framework and Gazebo simulator. There are many interesting documentation about them on internet, including their official webpages. In RoboticsAcademy we are using ROS1-Melodic and Gazebo-9 so far. In the next months we will migrate towards ROS1-Noetic and Gazebo-11, and later to ROS2.

Spnetic-5, 2021-03-29:
Okay, I'm familiar with ROS and Gazebo as well. But I'm unable to figure out how my approach should be towards this issue. I'll figure it out soon and come up with the solution!

jmplaza, 2021-03-29:
As an illustrative example, check how the sky is introduced in this case: https://uuvsimulator.github.io/packages/uuv_simulator/docs/tutorials/seabed_world/#creating-a-new-gazebo-world

Spnetic-5, 2021-03-29:
Got it, I will update soon!

predator4hack, 2021-03-29:
Please correct me if I'm wrong, I think that we'll have to add these lines of code:
`<scene>
      <sky>
        <clouds>
          <speed>12</speed>
        </clouds>
      </sky>
`
in `kobuki_1_reconstruccion3d.world` in order to get clouds In Gazebo. However, when I'm trying to launch the `kobuki_1_reconstruccion3d.launch`, the gazebo opens with the error `Could not open file[kobuki_1_reconstruccion3d.world]`.

jmplaza, 2021-04-03:
HI @predator4hack , you better submit the Pull-Request once you have tested :-). Then a JdeRobot committer will merge it or give you feedback. 

Clouds in "3D visual reconstruction" exercise are not desirable, as they may confuse the stereo matching algorithms. In addition,  it is a good policy to avoid mixing problems, one issue per problem :-)

trunc8, 2021-04-04:
Hi, please can one of the maintainers look at #808, in which I have tried to solve this issue? I would be keen to know if some more files were meant to be edited and learn more about the project architecture!

closed 2021-04-04 by dvalladaresv