JdeRobot/RoboticsAcademy #796: new RADI for ROS2

pankhurivanjani, 2021-03-30:
@shreyasgokhale, This will enable us to move forward with our amazon warehouse robot exercise plans

Jovian-Dsouza, 2021-03-31:
> @shreyasgokhale, This will enable us to move forward with our amazon warehouse robot exercise plans

Cool , I was also thinking of working on the same.

jmplaza, 2021-04-25:
This RADI will be numbered as RADI-4.x.y. The convention is RADI-2.x.y for RADIs with ROS-Melodic inside (and Gazebo 9); and RADI-3.x.y for RADIs with ROS-Noetic (and Gazebo 11) inside.

trunc8, 2021-07-01:
Solved and tested below after merging the PR and building from the parent branch.

[image]

closed 2021-07-01 by trunc8