JdeRobot/RoboticsAcademy #806: Migrate follow turtlebot to Robotics Academy v2.3

iamarkaj, 2021-04-03:
This PR #805 attempts to migrate the follow_turtlebot exercise to v2.3

However, this are few changes that are needed to be done:
1. It is currently using roombaROS from [CustomRobots](https://github.com/JdeRobot/CustomRobots), this needs to be fixed because using the same robot will reduce the difference between version 2.3 and version 2.1
2. Stability of the drone needs to be fixed 
3. More buttons must be added to control the ground robot

dgvsnl, 2021-04-03:
I want to contibute to this issue

pariaspe, 2021-04-06:
Done in #805. RADI 0.3.14 will include exercise dependencies.

Some tasks remaining:
- Add buttons to start, pause and reset turtlebot
- Improve gazebo world to look nicer
- Modify launch file to use turtlebot xacro from drone_assets package instead local duplicated file

Credits to @iamarkaj for the migration

pariaspe, 2021-04-08:
Some other tasks to do as @iamarkaj said [[1]](https://github.com/JdeRobot/RoboticsAcademy/pull/835#issuecomment-815394833):

1. Update developer launching instructions inside RoboticsAcademy/exercises/drone_cat_mouse/web-template/README.md and RoboticsAcademy/exercises/follow_turtlebot/web-template/README.md since it no longer uses gzweb
2. Update RoboticsAcademy/RoboticsAcademy.html to include follow turtlebot

trunc8, 2021-04-08:
> Some other tasks to do as @iamarkaj said [[1]](https://github.com/JdeRobot/RoboticsAcademy/pull/835#issuecomment-815394833):
>
> 1. Update developer launching instructions inside [...]

Task 2 solved in PR #840.
Additional questions for Task 1:
1) What would the updated developer launching steps be? Should it guide how to mount a local volume into the Docker container for testing and debugging?
2) In what scenario would a user check the README inside the web-template directories? We might have to make changes at fewer places if we only maintain the guidelines in the docs. What is your opinion on this @pariaspe?

pariaspe, 2021-04-08:
Hi @trunc8,

> Task 2 solved in PR #840.
> Additional questions for Task 1:
> 
>  1. What would the updated developer launching steps be? Should it guide how to mount a local volume into the Docker container for testing and debugging?

The developer launching steps are the instructions to launch the exercise without the manager.py, ussually by running different docker bashes.

>   2. In what scenario would a user check the README inside the web-template directories? We might have to make changes at fewer places if we only maintain the guidelines in the docs. What is your opinion on this @pariaspe?

READMEs are for developers, but yes, you are right, keeping instructions in different locations is a headeach.

jmplaza, 2021-04-08:
The launching instructions should be in the webpage for each exercise itself, on GitHub Pages. The users and developers should see (and update) them there. Maybe we should move the RADI-launch there...

dgvsnl, 2021-04-10:
hello sir,
So basically for each exercises, we have to add RADI-launch.. for directly
launching web-templetes.

Question-  were to add RADI,
-- directly to each exercise folder
                OR
-- Add separate folder were all the RADI launch files are...

trunc8, 2021-04-10:
Can you clarify what you mean by "adding RADI" and "adding RADI-launch"?

dgvsnl, 2021-04-11:
OKay, I m on it.

dgvsnl, 2021-04-11:
I mean, we want the launching instruction of each exercise on the github
pages directly, RIGHT!!

So were should i append installation information --- in readme.md file?

pariaspe, 2021-04-11:
Hello @dgvsnl, we already have the running instructions in github pages for each exercise

jmplaza, 2021-06-23:
The exercise is ready and works with RADI 2.4.3

closed 2021-06-23 by jmplaza